vhda/verilog_systemverilog.vim #2: struct typedef indent

NicoLugil, 2015-02-06:
It seems I can not get the correct indent when I use a struct typedef as below. It for example indents as:

class z;

    typedef struct { 
        real a;
    int b;             
    int c;               
    real d; } ts;
    ts s[];

    int unsigned cnt=0;

    function new();
        super.new();
    endfunction;

endclass;

vhda, 2015-02-07:
Thank you for reporting this bug.
Could you please check if indenting is working properly now?
Also report any regression you detect.

NicoLugil, 2015-02-09:
Thanks for the fix! It works better now, but there is still one weird thing, if the typedef if preceded by a comment //------

Then I get:

class z;
    
    // this is a comment
    // -----------------
    typedef struct {
    real a;
    int b;
    int c;
    real d; } ts; 
    
ts s[];
        
typedef struct {
    real a;
    int b;
    int c;
    real d;
} ts2;
    
ts2 t[];
    
int unsigned cnt=0;
    
function new();
    super.new();
endfunction;

endclass

vhda, 2015-02-09:
The indent code is the part that I least modified from the original code. Allow me to review this issue in more detail.
Also, I foresee various issues in multiline comments (`/* */`) given how the code is currently implemented. I'll look into this at a later time.

vhda, 2015-02-09:
This behavior should have been corrected by 6b7a4e013dec54c905166ded09ee127d3e2510e4.

NicoLugil, 2015-02-10:
Thanks! Works fine.

closed 2015-02-10 by NicoLugil