vhda/verilog_systemverilog.vim #4: indenting with constraints

NicoLugil, 2015-02-17:
This code does not indent as expected:

```
task run_phase(uvm_phase phase);
    
    assert(my_seq.randomize());
    my_seq.start(low_sequencer_h);

    assert(my_seq.randomize() with {Nr==6;});
my_seq.start(low_sequencer_h);

assert(my_seq.randomize() with 
{Nr==6; Time==8;});
my_seq.start(low_sequencer_h);

endtask
```

vhda, 2015-02-17:
Had to treat `assert` as an exception.
This special case should be fixed, but other similar situations might still exist.

NicoLugil, 2015-02-18:
Thanks for the fix. Works great now!

vhda, 2015-03-02:
Would the following indentation be acceptable for you?

```
assert(my_seq.randomize());
my_seq.start(low_sequencer_h);

assert(my_seq.randomize() with {Nr==6;});
my_seq.start(low_sequencer_h);

assert(my_seq.randomize() with
    {Nr==6; Time==8;});
my_seq.start(low_sequencer_h);
```

NicoLugil, 2015-03-02:
Yes that would be fine.

vhda, 2015-03-02:
Indenting of assertions should now be correctly implemented.

closed 2015-03-02 by vhda