vhda/verilog_systemverilog.vim #5: fork join indenting

NicoLugil, 2015-02-17:
fork - join with named begin/end block indenting seems to be broken:

```
task run_phase(uvm_phase phase);
    
    fork 
    begin : isolating_thread
        do_something();
    end : isolating_thread 
join

endtask
```

NicoLugil, 2015-02-18:
Does not seem to be related to the naming of the block. E.g. following stuff also indents weird:

'''
    fork 
    begin
    do_something();
end 
join

    fork begin
        do_something();
    end
join
'''

vhda, 2015-02-18:
I've identified the location where this happens, but I need to understand what was its full purpose to guarantee that I do not regress something.
I'm also starting to think that best solution would be to re-write the indentation script from scratch...

vhda, 2015-02-19:
Out of curiosity, would you mind testing this [addon][1] and see if the indent works better?
At the moment I am not working with SV, so it's more difficult for me to test it myself.

[1]: http://vim.sourceforge.net/scripts/script.php?script_id=4743 "systemverilog.vim : Indent & syntax script for Verilog and SystemVerilog "

NicoLugil, 2015-02-19:
Do you mean putting the content of that projects' indent file into this projects indent file, and see what happens? I could give that a brief try tomorrow.

vhda, 2015-02-19:
I was proposing to completely replace the verilog_systemverilog addon with other one.
But your solution should work as well, as long as you properly rename the file :)
Thanks in advance.

NicoLugil, 2015-02-20:
I briefly tried it, but it seems to be broken in many places.
Also if you look to the issues of that project, you already see a lot of indent issues.
The indent of this project is I think better quality.

vhda, 2015-02-20:
Thanks for testing this for me.

vhda, 2015-02-21:
This specific issue should be fixed.
I still need to review the indentation of functions, which is not perfect at the moment, but the fix will require some fundamental changes to the script.

NicoLugil, 2015-02-23:
Thanks for the update.

As a minor thing: when there is a space after the begin statement the indenting is not ok, but that is probably another issue.

closed 2015-02-23 by NicoLugil

vhda, 2015-02-23:
I already found and fixed the white space issue.

NicoLugil, 2015-03-02:
Is this included in the latest release?

vhda, 2015-03-02:
It was lost in a local branch, but I've just cherry-picked it for you :)