vhda/verilog_systemverilog.vim #6: Indent bug when a comment is inside module instantiation

ghilton319, 2015-02-27:
I was trying out your plugin (looks promising). I discovered that comments inside a module instantiation break indenting when there is a line that starts with a comment.

Example: The auto formatting indents the code as following:
<pre><code>`ifdef DO_THIS
	device d1 (
		.port (port[1]),
		// .port2(), comment
		.*
		);
	`endif</code></pre>
Should be:
<pre><code>`ifdef DO_THIS
	device d1 (
		.port (port[1]),
		// .port2(), comment
		.*
	);
`endif</code></pre>

However this formats correctly (comment is after port connection):
<pre><code>`ifdef DO_THIS
	device d1 (
		.port (port[1]), 	// comment
		.*
	);
`endif</code></pre>

Other examples demonstrating the indent bug:
<pre><code>`ifdef DO_THIS
	device d1 (
		.port (port[1]),
		/* comment before port connection on the same line */ .port2()
		.*
		);
	`endif</code></pre>
<pre><code>`ifdef DO_THIS
	device2 d2 (
		.out,
		.a,
		.b(B)/*,
		TODO  .c(C) port not implemented yet */
		);
	`endif</code></pre>

vhda, 2015-02-28:
Give me some time to analyze this issue in more detail.

vhda, 2015-03-04:
This should now be fixed.

ghilton319, 2015-03-04:
The `// .port2(), comment` and `/*comment*/ .port(),` is fixed. I'm still seeing the issue with multi-line comments (`/* \n */`). Examples:

    `ifdef DO_THIS
    	device d1 (
    		.port (port[1]),
    		/* comment line 1
    		* comment line2 */
    	       .port2(),
    		       .*
    	       );
           `endif

and

    `ifdef DO_THIS
        device2 d2 (
            .out,
            .a,
            .b(B)/*,
            TODO  .c(C) port not implemented yet */
            );
        `endif

vhda, 2015-03-04:
Issues with multi line comments, at the moment, are very difficult to fix because code only looks at 2 to 3 lines individually.

In the future I intend to use syntax regions to identify if lines are inside a multi line comment.

lewis6991, 2016-02-15:
Is this not fixed now?

vhda, 2016-02-15:
Just included more code from this issue in our test file.

Yes, everything appears to be correct! :)

vhda, 2016-02-15:
@lewis6991 Thanks once again for this great addition to **our** addon!

closed 2016-02-15 by vhda

lewis6991, 2016-02-15:
Your very welcome. :)