vhda/verilog_systemverilog.vim #9: Syntax highlighting disappears for 3+ digit time units

ghilton319, 2015-02-27:
Syntax highlighting disappears for the following examples:

* 100ns
* 2_0ps
* 3_000_000s

vhda, 2015-02-28:
Should be ok now.
Thanks for reporting.

vhda, 2015-03-02:
Please confirm if it working now.

ghilton319, 2015-03-02:
`123step` doesn't highlight. However when I look at the LRM, I only see `1step` every being used. Several simulators I have tried give compiling errors with any `\dstep` other than `1step`. In [IEEE Std 1800-2012](http://standards.ieee.org/getieee/1800/download/1800-2012.pdf) Annex A, Formal syntax only lists `1step` as a delay unit. I'm not sure if any `step` other then `1step` should be legal.

On another note, (maybe I'm a little to nit-picky) 1.2.3 should not highlight. I'm not even sure if this is fixable. Basic number formatting appears to be inherited.

Overall it is an improvement. I'll await feedback from you on the >`1step` and `1.2.3` before I close the issue. Thanks.

vhda, 2015-03-02:
I confirm the `1step` case. In fact, I have the fix prepared on a separate regular expression optimization commit where I replaced the match with a keyword.

I'll take a look at the 1.2.3 case, though. Should have thought about that yesterday! :)

vhda, 2015-03-02:
You will still see the `1.2.3` highlighted, but if you look carefully you'll see that it is identified as two separate number. I've considered the possibility of trying to avoid this, but I think the extra complexity in the syntax highlight might not be worth the effort. This is specially true if one considers the extra processing time of the regular expression in bigger files.

ghilton319, 2015-03-02:
Thanks for confirming `1step`.
I assumed `1.2.3` was identified as two separate numbers as well. I agree, the extra complexity in the syntax highlight might not be worth the effort. Probably something the core vim syntax detection should handle.
Thanks for the fix.

closed 2015-03-02 by ghilton319

vhda, 2015-03-02:
Forgot to mention that I included the `1step` fix, but I guess you noticed it ;)