vhda/verilog_systemverilog.vim #11: # moves to column 0 if smartindent is enabled

ghilton319, 2015-03-02:
I noticed that when stating a new line with `#` and smartindent is enabled, the line will un-indent back to column 0. Formatting with `==` or `gg=V` corrected the indenting, but the extra step is not desired.

Poking around the vim documentation I found [this](http://vimdoc.sourceforge.net/htmldoc/options.html#%27smartindent%27):
>When typing '#' as the first character in a new line, the indent for that line is removed, the '#' is put in the first column.  The indent is restored for the next line.  If you don't want this, use this mapping: ":inoremap # X^H#", where ^H is entered with CTRL-V CTRL-H.

Adding `au FileType verilog_systemverilog inoremap # X^H#` to my `~/.vimrc` is one solution. I am more in favor of add the below to `indent/verilog_systemverilog.vim` (or another file) so other users with smartindent will not run into the same issue.

    " inoremap if smartindent is on (^H is CTRL-V CTRL-H)
    inoremap # X^H#

From my tests, the change only effects the intended verilog/systemverilog files. c/cpp files retain the column shift functionality.

Note: don't copy-past `inoremap # X^H#`, the `^H` need to entered as `CTRL-V CTRL-H` in vim

vhda, 2015-03-03:
I don't think you should be using `smartindent` in Verilog/SystemVerilog.
What are you trying to accomplish that is not done already by `autoindent`?

vhda, 2015-03-04:
[This](http://vim.wikia.com/wiki/Indenting_source_code) is an interesting read, in particular section [File-type based indentation]( http://vim.wikia.com/wiki/Indenting_source_code#File-type_based_indentation).

ghilton319, 2015-03-04:
`smartindent` was enabled for some non-Verilog/SystemVerilog files. For want ever reason it is as a global and not a specific file time; I inherited the .vimrc file. I'm not exactly sure what file types need `smartindent`. I other users that point to the same file so I'm less inclined to make a global change without knowing the impact.

Yes, can can easily fix my local .vimrc. I not know how common `smartindent` is used with other vim users. It should not be a fix for me, but for the average vim user using this code. If `smartindent` is common, then it should be intelligently disabled within verilog_systemverilog.vim. If not, then close this as a "Won't Fix" to be a reference to others who have a similar complaint in the future.

vhda, 2015-03-05:
What if I raise a warning/error when loading this plugin if `smartindent` is defined?

ghilton319, 2015-03-05:
Raising a warning is a good idea. Error should only be if it breaks something and that is not this case.

vhda, 2015-03-05:
Ok, I'll implement this as soon as I have some time :)

vhda, 2015-03-07:
If you have any problems updating the plugin please reinstall it. Sorry about that.

ghilton319, 2015-03-09:
Did you implant the warning or accidentally comment on the wrong issue? I don't see a behavior change.

vhda, 2015-03-09:
I still did not worked on that.
But I was forced to re-write the complete git history of the project and used this issue as a way of alerting of that.

vhda, 2015-03-09:
Could you please check if the warning message is working correctly on your side?

ghilton319, 2015-03-11:
Interesting. The warning does not show up when I first open vim with a file (ex `% vim test.sv`) but does when I open or switch between files within vim (ex: `:e other.sv` , `:b test.sv`).

vhda, 2015-03-11:
Is it possible that you are setting `'smartindent'` through an `autocmd`?

ghilton319, 2015-03-11:
I'm not using `autocmd`. I used `set si` im my local `.vimrc`. If I commenting it out then I don't get the warning, so nothing else is enabling `smartindent`. 

To eval, I simplified my `.vimrc` to as bellow and my `~/.vim` only contains this project without modifications. Something might be suppressing the warning on first open. Not sure how. I get the same behavior if I run vim with absolute path (ex `% /usr/bin/vim test.sv`)

    set nu " show line-number
    set mat=3 " How many tenths of a second to blink
    set si " smartindent on
    "set noci " copyindent off
    set hls " highlight match search
    set sm " automatic matching braces
    set tw=0 " textwidth, wrap words over a certain length, 0==no-wrap
    set bs=2 " fix backspace on linux
    syn on " syntax format on

    "set expandtab       "Use softtabstop spaces instead of tab characters for indentation
    "set shiftwidth=2    "Indent by 2 spaces when using >>, <<, == etc.
    "set softtabstop=2   "Indent by 2 spaces when pressing <TAB>

    "set autoindent      "Keep indentation from previous line
    "set smartindent     "Automatically inserts indentation in some cases
    "set cindent         "Like smartindent, but stricter and more customisable
    
    "set foldmethod=syntax
    "let g:verilog_syntax_fold = 'all'

    filetype plugin indent on

vhda, 2015-03-11:
Found it!
Have to force a `:redraw` to guarantee that the message is correctly displayed during startup. Just hope I'm not abusing this command...
I'll submit the fix shortly.

vhda, 2015-03-11:
Please try now.

ghilton319, 2015-03-12:
I see the warning when starting vim with a systemverilog file now. thanks.

closed 2015-03-12 by ghilton319