vhda/verilog_systemverilog.vim #14: pure virtual functions indent wrong

NicoLugil, 2015-03-05:
This does not indent correctly:

```
virtual class base;

    pure virtual function void a(input int unsigned N, ref t Data);
        pure virtual function void b(input int unsigned N, ref t Data);
            pure virtual function void c(input int unsigned N, ref t Data);

            endclass;
```

vhda, 2015-03-05:
Please try now.

NicoLugil, 2015-03-06:
I dont know if this is new or was already like that but it seems all extern and pure virtual functions are now no longer indented in their class:

```
virtual class base;

extern function void x(input int unsigned N, ref t Data);
extern function void y(input int unsigned N, ref t Data);

pure virtual function void a(input int unsigned N, ref t Data);
pure virtual function void b(input int unsigned N, ref t Data);
pure virtual function void c(input int unsigned N, ref t Data);

endclass;
```

vhda, 2015-03-06:
I screwed up in the last fix. Sorry about that.

vhda, 2015-03-07:
If you have any problems updating the plugin please reinstall it. Sorry about that.

NicoLugil, 2015-03-08:
I can not try right now, but would there be a particular reason for that?
So far I have always just downloaded and extracted the zip.

vhda, 2015-03-08:
I was assuming that you were using a plugin manager like Vundle or Pathogen. But in your case there is no problem.

vhda, 2015-03-09:
BTW, I hope this issue is now fixed. Could you please confirm?

NicoLugil, 2015-03-10:
Yes this works fine! Thanks!

closed 2015-03-10 by NicoLugil