vhda/verilog_systemverilog.vim #15: assert with else clause on next line indent

NicoLugil, 2015-03-12:
This indents not as expected:

```
task run_phase(uvm_phase phase);

    assert(out>0)
        else $warning("xxx");
        $display("Hi");

    endtask
```

NicoLugil, 2015-03-12:
Actually with the else clause on the same line it also does something weird:

```
task run_phase(uvm_phase phase);

    assert(out>0) else $warning("xxx");
    $display("Hi");
$display("Hi");

    endtask
```

NicoLugil, 2015-03-13:
I am not sure this is easy to solve, but I often use a define for assertions, would be great if that would also indent correctly:

```
    // wait [1:5] clock cycles
    `assert_fatal(my_ll_wait_clk_seq.randomize() with {Nr inside {[1:5]};})
my_ll_wait_clk_seq.start(lowlevel_sequencer_h);

// reset [1:5] clock cycles
`assert_fatal(my_ll_do_reset_seq.randomize() with {Nr inside {[1:5]};})
    my_ll_do_reset_seq.start(lowlevel_sequencer_h);
```

vhda, 2015-03-24:
First part should be fixed.
Second one is tricky... so give me a bit more time to look into it.

NicoLugil, 2015-03-25:
I can confirm the first thing is working now! Thanks!

vhda, 2015-04-17:
Could you please check the second part?

NicoLugil, 2015-08-21:
Yes seems to work fine! Thanks

closed 2015-08-21 by NicoLugil