vhda/verilog_systemverilog.vim #23: OmniComplete doesn't work

lewis6991, 2015-10-28:
Looks to me like there a syntax error in the omni complete function. When I try to use omni, vim gives me the following error:
```
Error detected while processing function verilog_systemverilog#Complete:
line 119:
E583: multiple :else:	else
```

File used on:
```systemverilog
class b;
   bit field1;
   bit field2;
endclass : b

class a extends uvm_sequence;
    b b_inst;

    function update();
        b_inst.[omni complete here]
    endfunction : update
endclass : a
```

vhda, 2015-10-28:
I've just pushed the fix in https://github.com/vhda/verilog_systemverilog.vim/commit/600a102b05e53091462926874c9559014657d7bc, which was laying around in my work repository.

lewis6991, 2015-10-28:
I made the same change myself and confirm does work.

closed 2015-10-28 by lewis6991

vhda, 2015-10-28:
Perfect!
I am aware that omni-completion still has some limitations, so feel free to submit enhancement requests for features you would find important.
Pull Requests are also welcome :)