vhda/verilog_systemverilog.vim #31: Testing folds

lewis6991, 2015-10-30:
Is there any way of automatically testing folds or is it done manually at this point. If it's manual then I think I have a way of testing it automatically.

vhda, 2015-10-30:
It's all tested manually.
In fact, even today I asked a question in stackexchange asking for ideas on how to test omni completion automatically.
For indenting a simple ex command should be sufficient, but I still did not look into that.

lewis6991, 2015-10-30:
Check out 3bb677cec81091347056589ec70bf50aa1e1d94c and tell me what you think? It isn't perfect but it better than nothing.

vhda, 2015-10-31:
The only other alternative I can think of would be to use FastFold plugin and `:mkview`.
But your solution is more maintainable in my opinion.

lewis6991, 2015-10-31:
I just conjured up this.
```shell
vim folding.v -c "if verilog_systemverilog_test#TestFold() | cq | else | q | endif"
```
Running this with 8129166a7438606b0adb3cfc064a678e87a00609 will cause the exit code to match the result of the test.

vhda, 2015-11-02:
Please check my "travis" branch, which was built on top of your commit.
This is the [result](https://travis-ci.org/vhda/verilog_systemverilog.vim/builds/88816877).
I still want to re-review everything to see if there are any optimizations that can be done.

lewis6991, 2015-11-03:
Very nice. I've never used travis-ci before. Can it run from command line or does it require being used in the browser? What kind of optimizations do you mean, to the test or the travis infrastructure?

vhda, 2015-11-04:
travis-ci is a automatic regression service. Whenever a new commit us pushed into the repository or a Pull Request is created it will run all the test cases and display the result.

By "optimizations" I meant some code reorganization and improvement of this plugin's test infrastructure. You should now be able to see what (I hope) is the final structure.

vhda, 2015-11-05:
I think that this issue is resolved.
Feel free to reopen if if you feel otherwise.

Thanks for providing this idea and create the Vim script that is fundamental for this to work.

closed 2015-11-05 by vhda