vhda/verilog_systemverilog.vim #32: Folds not working properly got some if statements

lewis6991, 2015-11-02:
The following does not properly fold:

```systemverilog
if (cond1) begin
    do1(); 
end else if (cond2) begin
    do2();
    do3();
    do4();
end else begin
    do5();
end
```
And further folds even worse:
```systemverilog
if (cond1) 
    do1();
else if (cond2) begin
    do2();
    do3();
    do4();
end else begin
    do5();
end
```

vhda, 2015-11-09:
Please check my new branch `syntax/advanced-block-folding`, which includes the following commit: https://github.com/vhda/verilog_systemverilog.vim/commit/efed0a0ec94cd95137e2129d0932c00a94204b30.
This makes the folding test fail because of a small difference in how the end of the blocks is determined, but up until now I was not able to find any better solution. Any suggestions are welcome!

If this difference in folding persists I'll most likely add this a special "block_nested" option, to allow a precise folding with the disadvantage of not supporting nested blocks.

lewis6991, 2015-11-09:
This commit really messes with the indenting.

lewis6991, 2015-11-09:
I just took a look at the folding behaviour for similar constructs in other languages (c, c++, java) and none of then have a good solution to this. I have a feeling there may not be a good (if any) solution to this and maybe it might be best to just stick with the old behaviour. As long as the you put ```end``` on its own line, everything works fine.

vhda, 2015-11-17:
This behavior can be improved by adding "block_nested" string to `g:verilog_syntax_fold` option.

closed 2015-11-17 by vhda