vhda/verilog_systemverilog.vim #41: Indent script does not work well with uvm macros

lewis6991, 2015-11-11:
Because the indent script uses ```);``` to determine indents, uvm macros don't indent properly.

For those who do not know, UVM is an infrastructure/library for SystemVerilog to provide a structured verification methodology. UVM uses `define macros heavily. For example the error reporting macro is:
```systemverilog
`uvm_info("TAG", "message", UVM_MEDIUM)
```
The awkward thing about these macros is they do not require a ```;``` at the end of them and some simulators actually provide warning if you do. Also in some cases providing the ```;``` can cause an error:
```systemverilog
if (condition)
  `uvm_info("TAG", "message1", UVM_MEDIUM);
else
  `uvm_info("TAG", "message2", UVM_NONE);
```
This will cause an error because the simulator will see ```;;``` after the first ```if``` and will put the ```else``` out of context.

I understand that this is a complex issue so I accept that a solution to this may be infeasible.

Any solution would have to detect a ```)``` and then check if it belongs to a macro or not.

vhda, 2015-11-17:
I've included a section to the indent test containing the code that you provided and it appears to be correct. Could you please review this issue and provide an example that makes indenting fail?

Thanks!

lewis6991, 2015-11-17:
The problem I have is when a following statement does not de-indent when a macro follows an ```else``` on the same line. Examples are:
```systemverilog
assert property (prop1)
else `uvm_fatal("TAG", "Assertion failed.")

do_something(); // This doesn't indent properly

assert property (prop1)
else
    `uvm_fatal("TAG", "Assertion failed.")

do_something(); // This indents fine

if (condition)
    something();
else `uvm_fatal("TAG", "This is invalid.")

do_something(); // This doesn't indent properly

if (condition)
    something();
else
    `uvm_fatal("TAG", "This is invalid.")

do_something(); // This indents fine
```

vhda, 2016-02-12:
Fixed in #48 by yourself.

closed 2016-02-12 by vhda