vhda/verilog_systemverilog.vim #49: Odd indentation when using vlog 2001 parameterized instantiation/declaration

xguerin, 2016-01-21:
What should be:

```verilog
module MyModule #(
  parameter A = 1,
  parameter B = 2
)(
  input I,
  output o
);
```

is indented as follows:

```verilog
module MyModule #(
  parameter A = 1,
  parameter B = 2
  )(
    input I,
    output O
  );
```

And throws off all subsequent indentation. The same goes with instantiation:

```verilog
MyModule #(
  .A (1),
  .B (2)
) Module_1 (
  .I (...),
  .O (...)
);
```
is indented as follows:

```verilog
MyModule #(
  .A (1),
  .B (2)
  ) Module_1 (
    .I (...),
    .O (...)
  );
```

lewis6991, 2016-01-21:
This should be fixed in https://github.com/vhda/verilog_systemverilog.vim/pull/48 which is still under review. It is a complete rewrite of the indent script which addresses a lot of issues with the current.

If you could test the new script then that would be much appreciated: https://github.com/lewis6991/verilog_systemverilog.vim

xguerin, 2016-01-21:
It seems to be working like a charm, except maybe for expression that span over multiple lines, like this one:

```verilog
wire Val_IP  = !In_Pkt_S_Bus_enf || ((Pls_Don || ResVal) && (Pls_Res || ResFnd));
```

If I split that line in two (at the || statement for instance), it is indented as follows:

```verilog
wire Val_IP  = !In_Pkt_S_Bus_enf ||
((Pls_Don || ResVal) && (Pls_Res || ResFnd));
```

Instead of:

```verilog
wire Val_IP  = !In_Pkt_S_Bus_enf ||
                       ((Pls_Don || ResVal) && (Pls_Res || ResFnd));
```

lewis6991, 2016-01-21:
Thanks for reporting this. The indenting can get a bit unpredictable if the code isn't wrapped inside a context block, e.g. `begin end`, `function endfunction`.

The indent is designed to be as:
```systemverilog
wire Val_IP  = !In_Pkt_S_Bus_enf ||
    ((Pls_Don || ResVal) && (Pls_Res || ResFnd));
```
I'll try to get that fixed soon.

To produce the indent you specified would be quite difficult to implement reliably. However I would like to eventually add that feature, but probably not in the near future. The new script is significantly slower than the old one so the focus now is to fix bugs and improve performance.

vhda, 2016-01-21:
Carefull, you have different possibilities:

```verilog
wire Val_IP  = !In_Pkt_S_Bus_enf ||
               ((Pls_Don || ResVal) && (Pls_Res || ResFnd));
```

or:

```verilog
wire Val_IP  =
    !In_Pkt_S_Bus_enf ||
    ((Pls_Don || ResVal) && (Pls_Res || ResFnd));
```
So you need to look at what is in front of the `=`sign.

lewis6991, 2016-01-21:
Exactly. Quite difficult to get right. Currently the script is designed to indent as the latter.

vhda, 2016-02-12:
Partially fixed by PR #48.
Still missing the assignment indentation.

vhda, 2016-02-13:
PR #53 should fix the assignments, allowing us to close this issue.

vhda, 2016-02-15:
Fixed in PR #53

vhda, 2016-02-15:
Fixed in #53

closed 2016-02-15 by vhda