vhda/verilog_systemverilog.vim #52: .v files detected as verilog

xguerin, 2016-02-10:
Sorry for the n00b question, but I just realized that my .v files are still recognized as "verilog" files and not  "verilog_systemverilog". Any reason why the default filetype file would take precedence ?

vhda, 2016-02-10:
What does `:verbose set filetype` return?

xguerin, 2016-02-10:
filetype=verilog
        Last set from /usr/share/vim/vim74/filetype.vim

vhda, 2016-02-10:
What about `:autocmd * *.v`?

xguerin, 2016-02-10:
```vim
:autocmd * *.v
--- Auto-Commands ---
filetypedetect  BufNewFile
    *.v       setf verilog
BufNewFile
    *.v       setfiletype verilog_systemverilog
filetypedetect  BufRead
    *.v       setf verilog
BufRead
    *.v       setfiletype verilog_systemverilog
```

I don't have any particular au in my rc.

vhda, 2016-02-10:
Could you show me the contents of `ftdetect/verilog_systemverilog.vim` in your installation?

xguerin, 2016-02-10:
```vim
" Vim filetype plugin file
" Language:     SystemVerilog (superset extension of Verilog)

au! BufNewFile,BufRead *.v,*.vh,*.sv,*.svi,*.svh        setfiletype verilog_systemverilog
```

vhda, 2016-02-10:
As you may have noticed, `filetype` is set through auto-commands. You have the original auto-command as defined in `/usr/share/vim/vim74/filetype.vim` **and** the one set by this plugin. In my environment I only get:

```vim
--- Auto-Commands ---
filetypedetect  BufNewFile
    *.v       setfiletype verilog_systemverilog
filetypedetect  BufRead
    *.v       setfiletype verilog_systemverilog
```

And this happens even when I don't load my vimrc, so it shouldn't be related with any of my configurations.

What vim version are you running?

Could you also try the following?

1. Start vim with `vim -u NONE path/to/file.v`
2. `:set nocompatible`
3. `:set runtimepath+=~/.vim/bundle/verilog_systemverilog.vim` (adjust path to your environment)
4. `:filetype on`
5. `autocmd * *.v`

xguerin, 2016-02-10:
VIM - Vi IMproved 7.4 (2013 Aug 10, compiled Aug 20 2015 10:38:14)

```vim
filetypedetect  BufNewFile
    *.v       setf verilog
BufNewFile
    *.v       setfiletype verilog_systemverilog
filetypedetect  BufRead
    *.v       setf verilog
BufRead
    *.v       setfiletype verilog_systemverilog
```

vhda, 2016-02-10:
What is the "Included patches" range?

xguerin, 2016-02-10:
Included patches: 1-207, 209-801, 803-808, 810-827

I tried with vim's HEAD, same result.

xguerin, 2016-02-10:
Ok, I found the problem. It's not even remotely related to this plugin. It's an issue with how "Plug" enable its plugins.

vhda, 2016-02-10:
Just compiled vim from source and am still unable to replicate that behavior.

Are you 100% you followed the 5 steps above? According to `:help autocmd-remove` the line

```vim
au! BufNewFile,BufRead *.v,*.vh,*.sv,*.svi,*.svh        setfiletype verilog_systemverilog
```

should remove any existing auto commands associated with those patterns and add the new one isolated.

vhda, 2016-02-10:
I'm also using vim-plug... could you please provide more information, just in case I (or anyone else) has the same problem?

xguerin, 2016-02-10:
Sure. The culprit is that the plugin must be force-enabled like this:

```vim
Plug 'lewis6991/verilog_systemverilog.vim', { 'for': [ 'verilog_systemverilog' ] }
```

If not, for some reason the filetypes are not taken into account. Also, for those using indenline, indentline **must** be loaded first:

```vim
Plug 'Yggdroot/indentLine', { 'for': 'verilog_systemverilog'  }
Plug 'lewis6991/verilog_systemverilog.vim', { 'for': [ 'verilog_systemverilog' ] }
```

vhda, 2016-02-10:
Could I kindly ask you to comment out all plugins except for verilog_systemverilog.vim? Also, please remove the `{ 'for': ...}` part.

xguerin, 2016-02-10:
Sure:

```vim
filetype=verilog
     Last set from /usr/local/share/vim/vim74/filetype.vim
```

lewis6991, 2016-02-10:
I failed to reproduce this issue with the following vimrc:
```
set nocompatible

call plug#begin('~/.vim/plugged')
Plug 'lewis6991/verilog_systemverilog.vim'
call plug#end()
```
I also ran `PlugClean` and checked my `~/.vim` directory only contained `vim-plug` and `verilog_systemverilog`. My version of vim is macvim 7.4.1265.

xguerin, 2016-02-10:
It works fine with a clean ~/.vim : 

```vim
  filetype=verilog_systemverilog
        Last set from ~/.vim/plugged/verilog_systemverilog.vim/ftdetect/verilog_systemverilog.vim
```
I'll try to clean-up my own ~/.vim directory.

vhda, 2016-02-12:
@xguerin I'm curious to which file in your .vim folder was the culprit. Any news?

vhda, 2016-02-15:
@xguerin ? :)

xguerin, 2016-02-15:
Sorry, I haven't had much time to delve into that yet. I'll try to get to it shortly and I'll keep you posted

vhda, 2016-05-16:
I guess this problem stopped happening.

closed 2016-05-16 by vhda