vhda/verilog_systemverilog.vim #56: allow folding of module instances

SiliconKite, 2016-02-19:
Hello,

Currently it is possible to fold a number of constructs but as far as I know it is not possible to fold the text where verilog modules are being instantiated.

Would you consider this request?

Thanks anyway for the awesome plugin!

lewis6991, 2016-02-19:
I would like this too. We'll add it to our todo list. ;)

vhda, 2016-02-19:
Just did a quick test and as far as I can see we can't use `);` as an end pattern for syntax folding.

So, I think this is not going to be implementable unless we re-write all folding into a new `foldexpr` function.

lewis6991, 2016-02-19:
Whats preventing `);` from being an end pattern?

vhda, 2016-02-19:
Wait... the problem was in a missing `keepend` flag! :)

vhda, 2016-02-19:
@lewis6991 what do you think of #57?

lewis6991, 2016-02-19:
Would this break it?
```systemverilog
uvm_blocking_put_port #(trans) out;
```

vhda, 2016-02-19:
Yes, it does.

vhda, 2016-02-19:
Updated.

vhda, 2016-02-23:
@SiliconKite please note that the module declaration must contain a `(` or `#(` in the same line. As such the following will not fold:

```SystemVerilog
module_name 
instance_name(
     .port1 (net1),
     .port2 (net2)
 );
```

closed 2016-02-23 by vhda

SiliconKite, 2016-02-23:
Wow, this is some serious quick resolution!
Thanks a mill, guys :)

PR #57 (2016-02-23): Syntax: Add instance folding support
Fixes #56