vhda/verilog_systemverilog.vim #68: module not indenting?

NicoLugil, 2016-03-04:
I am hesitant to submit this issue as this pretty basic, and might be an error in my side.

```
module m (input clk);

import uvm_pkg::*;

initial begin
    $display("something");
end

endmodule
```

The content of the module does not seem to be indented?

lewis6991, 2016-03-04:
Try adding this to your vimrc.
```vim
augroup verilog_settings
    au!
    au FileType verilog_systemverilog let b:verilog_indent_modules=1
augroup END
```

NicoLugil, 2016-03-04:
yes this does the trick! Why isn't this default?

vhda, 2016-03-04:
There is a common rule that requires a single module declaration per file. When such rule is respected there is no gain in indenting the entire module: it starts in the first line and ends at the last line of the file.

This is specially relevant when using `shiftwidth=4/8` and `textwidth=80`, where available space becomes quite small.

The behavior was also the default in the original verilog_systemverilog plugin, so we're keeping it backward compatible (at least until the next major release).

closed 2016-03-04 by NicoLugil