vhda/verilog_systemverilog.vim #70: Incorrect indentation of multi-line condition.

lewis6991, 2016-03-08:
```systemverilog
cover property (
    a &&
        b &&
        c
);
```

vhda, 2016-03-08:
Let me also include the following scenario in this issue:

```systemverilog
        variable = (
            (a && b) ||
            (b || c)
            ) &&
            (c == d);
```

vhda, 2016-03-08:
Could you please @lewis6991? I'm trying to fix the "block_named" and "block_nested" folding, which have some highlight issues.

lewis6991, 2016-03-09:
No problem. I probably won't get around to it for a while though. The scenario you found looks like it will be a pain to fix. What would be correct indentation for it anyway?

vhda, 2016-03-09:
I'm open to suggestions, but in my opinion we should keep it indented as I pasted it above.

We could try to be smarter in regards to this indentation, but let's keep it simple for now.

lewis6991, 2016-04-09:
I think the indentation for the above should be:
```systemverilog

vhda, 2016-04-11:
I agree, that makes more sense than what I proposed.

lewis6991, 2016-05-18:
Looks like my original code insert has been fixed from other changes. Now just need to fix complicated variable assignment.

vhda, 2016-05-18:
I have a challenge for you... but I'll create a separate ticket for it.