vhda/verilog_systemverilog.vim #84: Last line of conditional assignment

lewis6991, 2016-04-27:
Current indent behaviour:
```systemverilog  
assign out = cond0 ? a :
             cond1 ? b :
             c;
```
Proposed indent behaviour:
```systemverilog  
assign out = cond0 ? a :
             cond1 ? b :
                     c ;
```

@vhda, what do you think about this.

vhda, 2016-04-27:
Personally I like it. But I propose that we add an option for it because it's possible that this goes against some coding guidelines.

lewis6991, 2016-04-27:
I agree. Lets add it to `g:verilog_disable_indent` as `conditional`.

vhda, 2016-04-27:
That's not ideal. For example, the coding guidelines I use require the assignment indentation but don't include the extra indentation you are now proposing. I can always manually delete the extra indent, but I would prefer a second variable.

(Sorry for the extra work)

lewis6991, 2016-04-27:
I'm proposing that `g:verilog_disable_indent='conditional'` simple disables the indent I want to implement, not assignment indents. The current behaviour makes no extra effort for conditionals.

Its important to differentiate the difference between `assignments` and `conditional assignments`.

vhda, 2016-04-27:
I misunderstood what you said. That seems perfect.

(I have a release this week... so I'm not really focused here, sorry).

vhda, 2016-04-27:
Just looking at bad code: take into account nested conditionals.

lewis6991, 2016-04-27:
Example?

vhda, 2016-04-28:
@lewis6991 I'm really sorry. Again, I did not look at this closely enough. Your original example already included nested conditionals. Hopefully, starting next week I'll be a bit more focused.

lewis6991, 2016-04-29:
No worries :)

lewis6991, 2016-05-04:
Fixed in #87

closed 2016-05-04 by lewis6991