vhda/verilog_systemverilog.vim #88: NC Verilog Errorformat Addition

goalposts, 2016-05-03:
I've been using the following with NCVerilog and UVM and it works reasonably well. UVM_FATAL does highlight during the summary report at the end, but the rest seems good. I would suggest test-driving and adding these to your existing toolset options.

" From https://github.com/ryanlee/vim/blob/master/vimfiles/compiler/ncverilog.vim
set errorformat+=%A%p\|,%Z%.%#:\ \*%t\\,%.%#\ \(%f\\,%l\|%c\):%m
" Ignore warnings (keep a copy looking for %t instead of W to define errors as errors)
set errorformat+=%-G%.%#\ *W\\,%.%#(%f\\,%l\|%c):\ %m
" Add abillity to parse verilog errors (%t matches 'E' or 'W' for type):
set errorformat+=%.%#\ *%t\\,%.%#(%f\\,%l\|%c):\ %m
" Assertions don't have the column information:
set errorformat+=%.%#\ *%t\\,%.%#(%f\\,%l):\ %m
"set errorformat=%.%#\ *%t\\,%.%#(%f\\,%l):\ %m
" Add abillity to parse NULL pointer errors:
set errorformat+=%E%>%.%#\ *E\\,%m,%C%.%#File:\ %f\\,\ line\ =\ %l\\,\ pos\ =\ %c,%Z%m
" Add abillity to parse assertion errors %.%# is like .* in regex:
set errorformat+=%.%#:\ *%t\\,%.%#\ (%f\\,%l):\ %m
" Add abillity to parse UVM errors:
set errorformat+=UVM_%tRROR\ %f(%l)\ %m
set errorformat+=UVM_%tRROR:\ %m
set errorformat+=%-GUVM_ERROR\ :%p0
"UVM_FATAL is an error that has no file associate with it.
" If we add it to errorformat, we lose visibility of "UVM_FATAL"
" This is an issue with the simulation report...
set errorformat+=UVM_FATAL\ %f(%l)\ %m
set errorformat+=%-GUVM_FATAL\ :%p0
set errorformat+=UVM_FATAL\ %m
"Ignore tool version information 'ncvlog(64): version......'
set errorformat+=-G^ncelab(64):%.%#
set errorformat+=-G^ncvlog(64):%.%#
set errorformat+=-G^irun(64):%.%#
set errorformat-=%f(%l):%m "This is subtractive as it catches the tool name as an error

vhda, 2016-05-05:
@goalposts I'd like to understand why all those entries are required. I'm particularly concerned about the `%-G` entries.

Would you mind that I include only the first errorformat and the basic UVM support in a temporary branch that you can try out and then you'll update this issue with examples of the special cases you find?

goalposts, 2016-05-05:
@vhda That sounds perfectly acceptable to me! The requirement for the -G entries come from the following:
set errorformat+=%-G%.%#\ W\,%.%#(%f\,%l|%c):\ %m ::: This is removing warnings from the log as it would be swamped otherwise. I defer to others here if warnings are required.
set errorformat+=%-GUVM_ERROR\ :%p0 ::: The report at the end of a successful simulation has UVM_ERROR : 0 , hence removing this from the global search for UVM_ERROR
set errorformat+=%-GUVM_FATAL\ :%p0 ::: As above
set errorformat+=-G^ncelab(64):%.%# ::: This one I can't quite see where the errorformat picked up ncelab(64) (or 32 actually), but this string removed them from the quickfix list. Similarly for the other tools. There's a good chance that the default errorformat for me has something which I'm undoing here.

All the best,

Dave

vhda, 2016-05-05:
Could you please check the `efm/ncverilog` branch?

vhda, 2016-05-06:
Just remembered that you need to do the following to enable the UVM error messages support:

```vim
let g:verilog_efm_uvm = 1
```

goalposts, 2016-05-06:
@vhda Yes, I spotted that. I've tried out the branch and added in a few more of the efm's as per the start of this thread. *I think* I have a pull request completed - first time using git would you believe?! I also just spotted that I may have inadvertently removed a fold marker...

I have a higher level question for you too regarding errorformats and the plugin's functionality. In my current flow I have log files with both gcc and ncverilog errors in there, and previously I had appended the ncverilog errorformats to the built-in errorformat rather than overwriting. Do you think adding in the option for additive errorformats would be possible? I guess this should be in another issue/enhancement request so feel free to move it.

vhda, 2016-05-06:
I can easily append it right after the UVM erroformat additions using a global variable.

Please make sure you include example error messages (you can omit confidential information related to file and signal names, obviously) that you want to catch/remove with the errorformat that you had to add.

vhda, 2016-05-14:
@goalposts could you please try my latest updates?
Thanks!

goalposts, 2016-05-24:
@vhda , apologies for the delay. I'm seeing the following being highlighted in the quickfix window with the latest updates:
TOOL:	irun	11.11-s111: Started on May 11, 1111 at 11:11:11 IST
The highlight returns with pipes rather than colons in the time reported (as a clue to the offending efm):
TOOL:	irun	11.11-s111: Started on May 11, 1111 at 11|11| 11 IST

That's the only issue I see at present...

vhda, 2016-05-24:
@goalposts are you sure you don't have extra `'errorformat'`entries on your side? I am unable to replicate the issue you are reporting.

vhda, 2016-06-20:
@goalposts could you please post the contents of your `'errorformat'` when those messages are reported in the quickfix window?

goalposts, 2016-06-21:
Hi @vhda , apologies for the delay again - was out of the office for a while. It looks like this is a case of PEBKAC as I must have loaded the log file with 'vim -q <logfile>' which circumvented the plugin's errorformat specifiers. I don't see the issue when I have the plugin loaded correctly and send the logfile to the location list separately.

vhda, 2016-06-24:
PR #111 includes the final commit with changes to `'errorformat'` to implement everything discussed in this issue.

vhda, 2016-06-27:
@goalposts Proper NCverilog suppoort is now part of the main branch. Please feel free to create a new issue in case you find any bug or limitation.

Thanks for your contribution to this project!