vhda/verilog_systemverilog.vim #99: Question Mark operator as part of search

TzachiNoy, 2016-05-15:
When searching a signal (by pressing * or **/**) vim doesn't find the cases it's followed immediately by question mark (**?**)

for example in the code below. pressing * when cursor is on b's declaration - won't find the b on the second row. And pressing / when cursor is on the second row won't take you back to it's deceleration, because it will look from \<b?\>.
```
   wire a,b,c,d;
   assign a=b?c:d;
```

I found that adding the following solved the problem:
```
   autocmd FileType verilog_systemverilog setlocal iskeyword-=63
   autocmd FileType verilog_systemverilog setlocal iskeyword-=?
```

lewis6991, 2016-05-16:
Nice catch? We'll keep the issue open until someone has time to open a PR to resolve this.

vhda, 2016-05-16:
@TzachiNoy thanks for reporting this issue. Feel free reopen it if you feel it is not fixed.

PR #100 (2016-05-16): Syntax: Remove `?` from keyword character list
Fixes #99
closed 2016-05-16 by vhda