vhda/verilog_systemverilog.vim #102: Special instance indentation support

vhda, 2016-05-18:
@lewis could you check how hard it would be to support the following indentation in instances?

```SystemVerilog
  my_module #(
    .PARAM1 (VALUE1),
    .PARAM2 (VALUE2),
    )
    u_my_instance(
      .port1 (wire1),
      .port2 (wire2)
      );
```

lewis6991, 2016-05-18:
What's the current and what is the desired?

vhda, 2016-05-18:
That's the desired.
Try copying to Vim and open a new line after the instance, you will see that the indentation will match the offset introduced in the instance.

lewis6991, 2016-05-18:
What config do you have with this?

lewis6991, 2016-05-18:
I currently get:
```systemverilog
  my_module #(
    .PARAM1 (VALUE1),
    .PARAM2 (VALUE2)
  )
  u_my_instance(
    .port1 (wire1),
    .port2 (wire2)
    );

logic a;
```

vhda, 2016-05-18:
You need to keep the indentation of the instance exactly as I copied above and only indent **after** the instance.

lewis6991, 2016-05-18:
Seems fine for me if I set `b:verilog_dont_deindent_eos` set.

vhda, 2016-05-18:
Ups! Needs an `ifdef`. I've edited my example above.

Sorry :)

lewis6991, 2016-05-18:
This won't be easy at all. I have a feeling there be may be a bug here around preprocessor statements.