vhda/verilog_systemverilog.vim #103: Preprocessor statements break indentation.

lewis6991, 2016-05-19:
````systemverilog
my_module u_my_instance(
  `ifdef TEST
    .port1 (wire1),
  `endif
  .port2 (wire2)
);

  logic a;
````

lewis6991, 2016-06-10:
Found another case which causes them to break:
```systemverilog
if (`SOME_TYPE(var))
  a = b; // Incorrect indentation.
```

vhda, 2016-07-01:
More examples.

In the following scenarios `my_function()` has one extra indentation level:
```SystemVerilog
  assign test = {
      3'b0
    , `ifdef SOMETHING 1'b1 `else 1'b0 `endif
  };
    my_function();
```
or:
```SystemVerilog
  assign test = {
      3'b0
    `ifdef SOMETHING
    , 1'b1
    `endif
  };
    my_function();
```

lewis6991, 2016-07-01:
I've been working on something for a while that should address this among a few other issues.