vhda/verilog_systemverilog.vim #104: Indent define blocks

lewis6991, 2016-05-24:
Current:
```systemverilog
`define macro_name(arg) \
always_comb begin \
    (arg) <= '0; \
end
```
Proposed:
```systemverilog
`define macro_name(arg) \
    always_comb begin \
        (arg) <= '0; \
    end
```

jeinstei, 2019-02-27:
I must admit to not being a fan of this by default. Would this be indented with reference to the current indent level, or only if not indented?

vhda, 2019-02-27:
Such indentation could always be disabled using `g:verilog_disable_indent_lst`.
All indentations are increments of the current level. That is, every line after `` `define`` would be indented one extra level when compared to the initial line, until the end of the `` `define``.

jeinstei, 2019-02-27:
Fair enough. Multi-line defines being indented makes sense. I misread the original intention.