vhda/verilog_systemverilog.vim #106: svp file extention

gombosh, 2016-06-19:
please add ".spv" files to filetype detection.
(pre precessor files, used in synopsys VIPs).

lewis6991, 2016-06-19:
Sounds fine with me, what do you think @vhda ?

vhda, 2016-06-20:
I did not find any existing filetype detection for .spv, so I see no reason not to add it to the list.

vhda, 2016-06-20:
@lewis6991 feel free to merge this PR if you agree with it.

vhda, 2016-06-20:
Wait... just noticed that @gombosh meant "svp" instead of "spv"...

vhda, 2016-06-20:
Updated title and description of original request according to what I think is the requested extension.
Created new PR with the correct fix.

closed 2016-06-20 by lewis6991

gombosh, 2016-06-21:
the requested extension is indeed ".svp", it stands for "system verilog protected" file (originally I thought it was "system verilog pre-processing", but I was mistaken).
it's funny since there's actually an "spv" verification language, but it's based on C++, so, not the case here :-)

vhda, 2016-06-21:
Hope it's now working for you :)

PR #109 (2016-06-20): Generic: Add extension .svp to file detection configuration
Fixes #106