vhda/verilog_systemverilog.vim #112: Incorret indent after comment block that has unmatched parenthesis on first line

ghilton319, 2016-07-07:
I've been experimenting with verilog-mode ([homepage](http://www.veripool.org/wiki/verilog-mode), [github page](https://github.com/veripool/verilog-mode)) AUTO_TEMPLATE feature and ran into an indenting issue with verilog_systemverilog. It will indent the proceeding lines of a multi-line comment block.
I discover this issue with AUTO_TEMPLATE, but other conditions can trigger it to, such as a commented out modue. 

Expected indent result:
```systemverilog
/* my_module AUTO_TEMPLATE (
.out (out[@]),
);*/
my_module inst_0(/*AUTOINST*/);
/*my_module dummy( // TODO
.*);
*/
```
Actual indent result:
```systemverilog
/* my_module AUTO_TEMPLATE (
  .out (out[@]),
  );*/
  my_modle inst_0(/*AUTOINST*/);
  /*my_module dummy( // TODO
    .*);
    */
```

The indent error occur when there is an open parenthesis on the first line of a comment block that does not have a matching closing parenthesis. If the open parenthesis is not on the first line or there is a matching closing parenthesis, then indenting works as expected. The following indent correctly:
```systemverilog
/*
my_module0 AUTO_TEMPLATE (
.out (out[@]),
);*/
my_modle0 inst_0(/*AUTOINST*/);

/* my_module1 AUTO_TEMPLATE ( .out (out[@]), );*/
my_modle1 inst_1(/*AUTOINST*/);

/* my_module2 AUTO_TEMPLATE
(
.out (out[@]),
);*/
my_modle2 inst_2(/*AUTOINST*/);
```

This appear to be an issue with indent/verilog_systemverilog.vim. I tried the the code with different file extensions ( .c .cc .cpp .java etc) and got the corrected result

lewis6991, 2016-07-07:
Thanks for reporting. I'm already in the middle of a large change so I'll try and incorporate a fix for this into that. Don't expect a fix any time soon though.

vhda, 2016-07-07:
@ghilton319 blasphemy! You've been using emacs!? :P

vhda, 2017-11-28:
@ghilton319 could you please check if this problem still occurs? Thanks!

ghilton319, 2018-01-04:
Looks okay

closed 2018-01-04 by ghilton319