vhda/verilog_systemverilog.vim #114: Tailing question mark of a number is not interpreted as conditional operator.

y-kim, 2016-07-14:
```verilog
3'b0?0, 3'b00?
```
Color of the first question mark is same to number, but that of the second is same to operator.

y-kim, 2016-07-14:
```vim
 syn match   verilogNumber "\(\<\d\+\|\)'[sS]\?[bB]\s*[0-1_xXzZ?]\+\>"
 syn match   verilogNumber "\(\<\d\+\|\)'[sS]\?[oO]\s*[0-7_xXzZ?]\+\>"
 syn match   verilogNumber "\(\<\d\+\|\)'[sS]\?[dD]\s*[0-9_xXzZ?]\+\>"
 syn match   verilogNumber "\(\<\d\+\|\)'[sS]\?[hH]\s*[0-9a-fA-F_xXzZ?]\+\>"
```
I think this problem came from Verilog syntax definition bundled in VIM 7.4 (related part is written above). It counts number until word boundary but the last question mask is not word by VIM.

vhda, 2016-07-14:
I think this is a feature, not a bug. The idea is to support the following statements:

```systemverilog
casez (ir)
  8'b1???????: instruction1(ir);
  8'b01??????: instruction2(ir);
  8'b00010???: instruction3(ir);
  8'b000001??: instruction4(ir);
endcase
```

lewis6991, 2016-07-14:
I think the both question should be the same color as number, however if there is a space then the question mark should match as an operator. 
````systemverilog
3'b000?0000? ?
````

vhda, 2016-07-14:
You're correct. I misunderstood the original report. My bad.

PR #115 (2016-07-16): Syntax/improve
Fixes #114
closed 2016-07-16 by vhda