vhda/verilog_systemverilog.vim #116: indent wrong (bug)

luoyanghero, 2016-08-07:
Expected:
`module tst(a,b);
input a;
output b;
endmodule
`
Actually:
`module tst(a,b);
    input a;
    output b;
endmodule`

vhda, 2016-08-07:
Hi @luoyanghero,

You should be able to use the following configuration to disable that indentation:

```Vim
let g:verilog_disable_indent_lst = "module"
```

Or using the command:

```Vim
:VerilogDisableIndentAdd module
```

For more information check the following help pages:

```Vim
:help :VerilogDisableIndentAdd
:help g:verilog_disable_indent_lst
```

The complete help text for this addon is available as following:

```Vim
:help verilog_systemverilog.txt
```

vhda, 2016-12-17:
Assuming fixed.

closed 2016-12-17 by vhda