vhda/verilog_systemverilog.vim #121: unique/unique0/priority syntax problem

r901042004, 2016-10-09:
When I use unique/unique0/priority + if/case/casez/casex ie. unique if.
The if/case/casez/casex statements are recognized as normal syntax.
How can I fix this problem?

lewis6991, 2016-10-09:
I'm not sure I understand your issue. Can you explain further?

r901042004, 2016-10-09:
Sorry, it's a highlight issue. I didn't set up "verilogConditional" of highlight option in my scheme, and it's solved now. And I found another problem.

In following systemverilog code:
```
always@(posedge clk or posedge rst)
begin
  if (rst)
    CS <= IDLE;
  else
    CS <= NS;
end
```
The "begin" and "end" are highlighted by "verilogStatement".
When I add "priority" before "if", the highlight of "end" will change to "Normal".
```
always@(posedge clk or posedge rst)
begin
  priority if (rst)
    CS <= IDLE;
  else
    CS <= NS;
end
```
Assume signal "clk" and "rst" are yellow (Normal highlight), and "begin" and "end" are purple (verilogStatement).
When I add "priority" before "if", "end" is change to yellow (Normal highlight).

r901042004, 2016-10-09:
I found that if I set `let g:verilog_syntax_fold_lst="all"` in ".vimrc", the problem will occur.
Normal highlight become Statement highlight when I set `let g:verilog_syntax_fold_lst="block_nested"` in ".vimrc", 
Setting `let g:verilog_syntax_fold_lst=""` in ".vimrc" could let highlights correct.

r901042004, 2016-10-09:
I found that if I set `let g:verilog_syntax_fold_lst="block,instance"` in ".vimrc", the problem will occur.
Normal highlight become Statement highlight when I set `let g:verilog_syntax_fold_lst="block_nested"` in ".vimrc", 
Setting `let g:verilog_syntax_fold_lst=""` in ".vimrc" could let highlights correct.

lewis6991, 2016-10-10:
@vhda implemented the `begin...end` folding so I will assign to him.

r901042004, 2016-10-10:
@lewis6991 Thank you.
I found another if-else and case-endcase highlight problems.
I set `let g:verilog_syntax_fold_lst=""` in ".vimrc", so it's might **not** a folding problem.

**If-else:**
When the cursor is on "else", the highlight is work.
[image]
It doesn't highlight when the cursor is on "if".
[image]

**If-else if-else**
When the cursor is on "else", "if" doesn't highlight.
[image]
It doesn't highlight when the cursor is on "else if".
[image]
[image]
It doesn't highlight when the cursor is on "if".
[image]

**case-endcase:**
When the cursor is on "endcase", the highlight is work.
[image]
It doesn't highlight when the cursor is on "case".
[image]

The "casez" and "casex" statement have the same problem.

vhda, 2016-10-10:
The first highlighting issue was already reported in issue #63 (by myself), but I still was not able to find a fix for it.

The other situation you referred is not related with this plugin, but with hl_matchit plugin. Nevertheless, I tried out the issues you reported and was unable to replicate. Please try to `:set updatetime=100` and see if it the behavior improves (note: I do not advise you to keep this setting, as it may make Vim slower). If this doesn't help, then please try disabling all other plugins, with the exception of verilog_systemverilog, matchit and hl_matchit.

Hope this helps!

r901042004, 2016-10-11:
@vhda I see. Thanks for your help.

r901042004, 2016-10-11:
I found the solution of hl_matchit plugin's problem.
Maybe you can give anyone who want to install the plugin a setting suggestion in [README.md](https://github.com/vhda/verilog_systemverilog.vim/blob/master/README.md).

1. The first issue is matchit plugin. Its developers didn't notice that Verilog has "else if" condition.
Type `:echo b:watch_words` in Vim, and we can find that it only has `\<if\>:\<else\>`.
Therefore, we can change it's value in .vimrc, it only changes when opening an verilog or systemverilog program:
```
autocmd FileType verilog_systemverilog let b:match_words='\<begin\>:\<end\>,`if\(n\)\?def\>:`elsif\>:`else\>:`endif\>,\<case\>\|\<casex\>\|\<casez\>:\<endcase\>,\<module\>:\<endmodule\>,\(else\s\+\)\@<!\<if\>:\<else\>\s\+\<if\>:\(`\)\@<!\<else\>\(\s\+if\)\@!,\<fork\>:\<join\(_any\|_none\)\?\>,\<function\>:\<endfunction\>,\<task\>:\<endtask\>,\<specify\>:\<endspecify\>,\<config\>:\<endconfig\>,\<specify\>:\<endspecify\>,\<generate\>:\<endgenerate\>,\<primitive\>:\<endprimitive\>,\<table\>:\<endtable\>,\<class\>:\<endclass\>,\<checker\>:\<endchecker\>,\<interface\>:\<endinterface\>,\<clocking\>:\<endclocking\>,\<covergroup\>:\<endgroup\>,\<package\>:\<endpackage\>,\<program\>:\<endprogram\>,\<property\>:\<endproperty\>,\<sequence\>:\<endsequence\>'
```
And now we can use % in normal mode to jump between if-else if-else condition.

2. The hl_matchit plugin can catch all condition now. We just need to change some global value to optimize it:
Enable hl_matchit on vim startup (1:enable 0:diable)
`let g:hl_matchit_enable_on_vim_startup=1`
Use speed level 2 otherwise hl_matchit sometimes doesn't highlight.
If this option makes vim slow, disable the plugin might a good choice.
`let g:hl_matchit_speed_level=2`
Decrease cursor waiting time.
`let g:hl_matchit_cursor_wait = 0.1`

vhda, 2016-10-11:
The matchit configuration is setup in [ftplugin/verilog_systemverilog.vim](https://github.com/vhda/verilog_systemverilog.vim/blob/master/ftplugin/verilog_systemverilog.vim), so you can apply the change there and propose it in a Pull Request.

As a side note, take into account that there is no formal "else-if" in Verilog and that you are using a trick to overcome that. Did you have the intention of also supporting the following examples?

```SystemVerilog
if (something1)
else
  if (something2)
  else

if (something1)
else // Otherwise, check for something2
  if (something2)
```

Even though I'm suggesting some plugins, I don't want to maintain documentation regarding their configuration because it will force me to track their updates to reflect on my own README.md.

And I need to create a PR for hl_matchit myself... just noticed that I have removed the dependencies on `g:hl_matchit_cursor_wait` locally to avoid making Vim slow. That's why when I want to make it faster I just need to configure `'updatetime'`.

r901042004, 2016-10-12:
The matchit plugin can't let cursor jump to correct position in following situation.
```
if (a)  // if 1
begin
  if (b)   // if 2
    statement;
end
else  //else 1
begin
  statement;
end
```
"If 1" jumps to itself, and "if 2" jumps to "else 1".
It must rewrite the plugin to correct it, so I give up.

For following code, I think that jump between "if1 - else1", "if2 - else if2 - else2", "if3 - else3" is acceptable for me.
We seldom need jump between "if1 - if2 - else if2 - else2 - else1" in most case.
```
always@(*)
begin
  if ( )  // if 1
  begin
    // (Maybe some statement is here)
    if ( )  //if 2
      ;
    else if ( ) // else if 2
      ;
    else // else 2
      ;
  end
  else //else 1
  begin
    if ( ) //if 3
      ;
    else //else 3
      ;
  end
end
```
BTW, IEEE standard has mention "if-else-if" statement in chapter 12.4.1 [here](http://standards.ieee.org/getieee/1800/download/1800-2012.pdf).
Furthermore, it must revise some script in matchit.vim. Otherwise, "if-else if" statement won't work. ("else if " can't jump back to "if").

vhda, 2017-01-23:
This should have been fixed, or at least improved, with PR #130.

vhda, 2017-03-06:
@r901042004 could you confirm if the original highlighting issue is fixed in the latest version?
Please note that I'm not referring to the highlight matching, which is out-of-scope of this plugin (if you agree?).

vhda, 2017-03-30:
Closing issue because there are no updates for an extended period of time.

closed 2017-03-30 by vhda