vhda/verilog_systemverilog.vim #126: Set error format in .vimrc

gogulfresh, 2016-11-03:
Hi Guys,
this is not really an issue but since I don't know where to ask I ask here... :)
How can I set the error format in my .vimrc? I don't like to type :VerilogErrorFormat every time I open vi...
Thanks for providing this great plugin!
Gogul

vhda, 2016-11-03:
Pick one of the following solutions by adding it to your vimrc:

1. compiler! something
2. autocmd FileType verilog_systemverilog compiler something

I prefer the second solution, which guarantees that `'errorformat'` is only set for Verilog/SystemVerilog files.

You may also need to configure the variables 'g:verilog_efm_level', 'g:verilog_efm_uvm_list', 'g:verilog_efm_quickfix_clean' and 'g:verilog_efm_custom'. Check their respective help pages for more information.

Hope this helps.

gogulfresh, 2016-11-10:
thank you! This is such a great feature! What is the difference between compiler and compiler! ?

closed 2016-11-10 by gogulfresh

vhda, 2016-11-10:
You can check the help for that using `:help :compiler`.

With the exclamation suffix the command applies to all buffers, while without it only applies to the current buffer.

vhda, 2016-11-10:
You're welcome :)