vhda/verilog_systemverilog.vim #128: Issue with indent and appending text in visual mode

justrajdeep, 2016-11-18:
Hi 
First i would like to thank you for this amazing plugin.

I am facing an issue, please let me know if it is a known one

I have code like below

            assign __intf.gsgrp_if.send = gsgrp_if.send;

        assign __intf.gsgrp_if.gsgrp_data.is_event   = gsgrp_if.gsgrp_data.is_event
        assign __intf.gsgrp_if.gsgrp_data.event_id   = gsgrp_if.gsgrp_data.event_id

I would like to add `;` in the last 2 assigns, i go to the 2nd assign and do the following key combination `C-vj$A;<ESC>`

This transforms the text to 

            assign __intf.gsgrp_if.send = gsgrp_if.send;

            assign __intf.gsgrp_if.gsgrp_data.is_event   = gsgrp_if.gsgrp_data.is_event;  
        assign __intf.gsgrp_if.gsgrp_data.event_id   = gsgrp_if.gsgrp_data.event_idvent;

because the line gets indented it messes up the text inserted. 

`gsgrp_if.gsgrp_data.event_idvent;` should be `gsgrp_if.gsgrp_data.event_id;`

Can you please take a look.

lewis6991, 2016-11-22:
I think this is caused because line 2 is indented incorrectly. I think the best way to achieve what you want is to not use visual block mode. `A;<ESC>j.` would achieve what you want just as well and is one character less. Other than that I can't see a clear way around this other than `set noautoindent`

vhda, 2016-11-22:
@lewis6991 I think this happens because `;` is part of `'indentkeys'` and so will trigger the call to `'indentexpr'`. Unfortunately, at the time when the indentation function is called only the first line ends in `;` and so the second line will not be indented.

The only thing we could do would be to assume the same indentation level as the previous line when the current does not end with `;`, but I think this would result in a big number of indentation failures. Which means there's no solution other than running `gv=` after appending the text in visual block mode.

justrajdeep, 2016-11-22:
@lewis6991 your mapping `A;<ESC>j.` would work for only 2 lines, if we have multiple lines it is not that helpful.

@vhda the issue is not that the 2nd line is not getting in indented, but the wrong text inserted in the 2nd line. i did not follow how `gv=` post text appending would solve the issue since the text appended would be wrong. building on your proposal if we do `=` (indentation of the block) and then proceed to insert text that might solve the problem.

vhda, 2016-11-22:
@justrajdeep sorry, I did not notice the extra characters before. That is very strange and would guess there is a bug in Vim. From our script point of view we only return the indentation value to Vim, which is then responsible for aligning the text.

justrajdeep, 2016-11-22:
@vhda with your plugin disabled vim works fine(tested it myself).
with the plugin enabled the 1st line gets indented and that messes up the text inserted in the visual block mode.

vhda, 2016-11-22:
@justrajdeep but does any of the lines get indented when you do that?

FYI, I've opened an issue in Vim for this vim/vim#1269. Let's see what they say.

justrajdeep, 2016-11-22:
@vhda no none of the lines get indented in that case.

vhda, 2016-11-22:
That's the reason you don't see the bug occurring. If the first line was indented new characters would appear in the second line in the same number as the indentation of the first line.

justrajdeep, 2016-11-22:
@vhda thanks. lets see what the vim guys respond.

vhda, 2017-02-24:
I'm receiving some updates on this... apparently the solution is not trivial.

Going to check a patch tonight to see if it fixes the issue.
More news as soon as I have them! :)

vhda, 2017-02-25:
The fix results in the following output:

```systemverilog
    assign __intf.gsgrp_if.send = gsgrp_if.send;
   
    assign __intf.gsgrp_if.gsgrp_data.is_event   = gsgrp_if.gsgrp_data.is_event;
assign __intf.gsgrp_if.gsgrp_data.event_id   = gsgrp_if.gsgrp_data.event_id;
```

I think this is a bit better. Take into account, though, that it is not possible to apply the indentation update to all modified lines.

vhda, 2017-10-19:
Fixed in vim/vim@e2e69e48134cbfdedea7802810932f8592705024 (patch 8.0.1041).

closed 2017-10-19 by vhda