vhda/verilog_systemverilog.vim #129: indentation issue "if" statement on 1 line

antoinemadec, 2017-01-11:
Hi,

First, thanks a lot for this rad plugin.
Then, the issue:
```
  if (cond) begin do_something; end
  do_something_else;
```
is wrongly indented as follows:
```
  if (cond) begin do_something; end
    do_something_else;
```
A one liner if is poor coding style to start with, but you don't choose how people write their code :p
Could you please take a look ?

Cheers,
Antoine

lewis6991, 2017-01-12:
We've had this issue raised before. It is something that we can fix but it would hurt performance enough for me to not want to change it.

> A one liner if is poor coding style to start with, but you don't choose how people write their code :p

I partly agree with you, however the issue is that SystemVerilog is an incredibly complex language syntactically and figuring out how to model every single case is near impossible. It's hard enough trying to indent sensible code!

With that said, I have been working on a complete rewrite of the syntax and indent scripts which fixes this case. Don't expect these to be finished anytime soon and @vhda may not agree with how I've implemented them so they may not get committed here. If you want to check it out then take a look at https://github.com/lewis6991/systemverilog.vim . It still has a lot of problems but it's good enough for my day-to-day use.

antoinemadec, 2017-01-12:
Thanks for the clear feedback.
I will take a look at your plugin, is based on https://github.com/nachumk/systemverilog.vim ?

Cheers,
Antoine

vhda, 2017-01-12:
Free time has been lacking lately and I've also been trying to fix the syntax (folding) issues.
Currently trying out a expression based folding, but it's just too damn slow.

@lewis6991 PR #119 is waiting on me, correct?
Is everything there similar to what you have under https://github.com/nachumk/systemverilog.vim?
Or is this PR no longer up-to-date?

lewis6991, 2017-01-12:
@antoinemadec no it is not based on that. My version is completely re-written based on what I've learnt from developing on this plugin.

@vhda https://github.com/lewis6991/systemverilog.vim is a complete re-write and isn't related to the PR. For the syntax, instead of just listing a gigantic list of keywords and patterns that match everywhere, I have defined regions and patterns to only match where they are syntactically valid. From my usage this makes things a lot quicker as it makes it easier for the Syntax engine as there are less patterns it needs to match against. This is then leveraged in the indent script that simply looks at the syntax region to immediately know the context of the line. It's not complete as I have only added syntax as I've needed it, however it completely passes the indent test of this plugin with some subjective differences.

I think it would be best to close the PR for now.

commit b293a96a (2017-04-24): Indent: Check for begin..end pairs before indenting

Fixes #129
PR #137 (2017-04-24): Indent: Check for begin..end pairs before indenting
Fixes #129
closed 2017-04-24 by vhda