vhda/verilog_systemverilog.vim #133: Syntax Highlighting Improvement

eolithic, 2017-03-11:
Since tags are already being used, could it be possible to highlight the classes (which have been already declared) at the time of instantiation or object creation? For eg.

Class a;
Endclass

Class b;
**_a_** a1;
function new();
   a1 = **_a_**::type_id::create("a1");
endfunction;
endclass

vhda, 2017-03-15:
I consider that functionality outside the scope of this plugin, because there are already many plugins doing that.

For example, two that I've just found:

* [Easytags](https://github.com/xolox/vim-easytags) - also maintains tag database automatically.
* [TagHighlight](https://github.com/vim-scripts/TagHighlight) - screenshots look nice, I should try it sometime.

Please do share which plugin you end up using.

vhda, 2017-03-30:
@eolithic did you end up testing any of these plugins?

vhda, 2017-04-07:
@eolithic any updates?

vhda, 2017-04-19:
Closing issue.
Please feel free to re-open if you feel that my answer was not sufficient.

closed 2017-04-19 by vhda