vhda/verilog_systemverilog.vim #134: Autocmd Filetype verilog not executing

jvmCarneiro, 2017-03-23:
Previous to the plugin installation I used the following augroup for verilog:
```
" Verilog setup
" Function to set verilog indentation
function! Vindent()
	setlocal tabstop=3
	setlocal softtabstop=3
	setlocal shiftwidth=3
	setlocal expandtab
endfunction

" Function to set verilog line wrapping
function! Vwrap()
	setlocal textwidth=80
	setlocal formatoptions+=t
endfunction

augroup verilog_autocmds
	au!
	" Applies functions automatically
	autocmd FileType verilog call Vindent()
	autocmd FileType verilog call Vwrap()
	" Highlights line limit at 80 char
	autocmd FileType verilog highlight OverLength ctermbg=DarkGrey ctermfg=White guibg=DarkGrey
	autocmd FileType verilog match OverLength /\%81v.*/
augroup END
```
However, none of those autocmd execute after installation. Should I change the FileType argument or something? I also use:
```
filetype plugin indent on
syntax on
```

vhda, 2017-03-23:
Olá João :)

This plugin implements a new Vim filetype called `verilog_systemverilog`, which means that your `FileType` events for `verilog` must now be modified to use `verilog_systemverilog` instead.

I'm inheriting the filetype defined by the plugin I used as a base for the current version you now have available and while I've considered many times to rename it, I simply decided that it would break too much the backward compatibility with previous versions.

Kind regards,
Vitor

jvmCarneiro, 2017-03-23:
Awesome, all's well now. A simple `:set filetype?` would've shown me that.

Keep it up Vitor, obrigado!

closed 2017-03-23 by jvmCarneiro