vhda/verilog_systemverilog.vim #135: some case .sv intend not right

luoyanghero, 2017-04-11:
This is in a .sv file, when I type 'extern function new(string name, virtual rdy_need_intf_v2 intf)' and then ';', it shows follow:
class rdy_need_slave_driver_v2;
    randc int l_cnt;
    randc int h_cnt;
    string name;
    virtual interface rdy_need_intf_v2 vif;

        extern function new(string name, virtual rdy_need_intf_v2 intf);
endclass : rdy_need_slave_driver_v2;

But I wish like this:
class rdy_need_slave_driver_v2;
    randc int l_cnt;
    randc int h_cnt;
    string name;
    virtual interface rdy_need_intf_v2 vif;

    extern function new(string name, virtual rdy_need_intf_v2 intf);
endclass : rdy_need_slave_driver_v2;

I try to delete line 'virtual interface rdy_need_intf_v2 vif;', I typed 'extern function new(string name, virtual rdy_need_intf_v2 intf)' and then ';', it shows follow:
class rdy_need_slave_driver_v2;
    randc int l_cnt;
    randc int h_cnt;
    string name;

    extern function new(string name, virtual rdy_need_intf_v2 intf);
endclass : rdy_need_slave_driver_v2;

Question: Is it a bug, or do I not setting right?

vhda, 2017-04-11:
This is a bug in the way interfaces are being identified. Basically, `virtual interface` is being identified as the start of an interface declaration, which results in an extra indentation level.

While I could add a quick patch to overcome this limitation, please give me some time to come up with a more generic (and elegant) solution.

luoyanghero, 2017-04-13:
Thanks a lot. I am waiting your results.

commit 459f2e17 (2017-04-19): Indent: Remove indentation of virtual interfaces

Fixes #135.
closed 2017-04-19 by vhda