vhda/verilog_systemverilog.vim #141: syntax highlight is lost on going to end of line in long line after patch v8.0.0646

justrajdeep, 2017-06-27:
Hi 

With the latest vim change [v8.0.0646](https://github.com/vim/vim/commit/0946326580e6f034fe2c88d041407ea0fde980ab) the syntax highlight is lost if i have a long line.

To reproduce:

have a long line (which does not fit the window) in an sv file. 
set `nowrap`
go to the end of line pressing `End`
syntax highlight is lost, which can be revived by pressing `Ctrl-l`

Can you please take a look?

Thanks in advance.

vhda, 2017-06-28:
At the moment I don't have that Vim version available. Could you please check if you have similar issues with other syntax types? To confirm if this is specific to this plugin.

justrajdeep, 2017-06-28:
Hi 

I did test out some other files and faced similar syntax issues in a few other plugins as well.

with `perl`: issue is there, goes away when i disable [vim-save-cursor-position](https://github.com/wellsjo/vim-save-cursor-position.git)

with `python`: no issue

with `cpp`: issue is there which goes away when i disable [vim-cpp-enhanced-highlight](https://github.com/octol/vim-cpp-enhanced-highlight.git)

with `csh`: no issue

Thanks

vhda, 2017-06-30:
From what you say, this should be a Vim issue.
But before telling you to open an issue in vim's Github, I would first recommend that you start by disabling all your plugins using `vim -u NONE -U NONE` and see if the default syntax files show the same error.

justrajdeep, 2017-06-30:
Hi 

I have tried it already with all plugins disabled. The default syntax files dont show any issues. Thant is why I concluded that it is not a vim issue.

Thanks

vhda, 2017-06-30:
Since I did not modify the plugin, I can only assume there was a change in Vim that resulted in this behavior. Do you know which version you had before, so that I can try to skim through the changes to see if I find anything that makes sense.

justrajdeep, 2017-06-30:
Hi

i have mentioned the vim version change that started this issue in the bug description. You can just look at that. Please let me know if u need any other info.

BTW: I am using gvim, on RHEL6, x86_64

Thanks

vhda, 2017-06-30:
I'm sorry but I am unable to replicate this issue.
You will need to understand which plugins are originating this behavior, but as far as I can tell it's not related with my plugin.
Note: I tested in RHEL 6.5, x86_64 in both vim and gvim (GTK2).

justrajdeep, 2017-06-30:
Ok .. i will check more and get back to you.

vhda, 2017-08-19:
Any updates @justrajdeep ?

justrajdeep, 2017-08-21:
hi vhda
i have not been able to rootcause it. If you want you can close this issue now. Once i have something concrete i will post and reopen the issue again.

vhda, 2017-08-24:
I'll close the issue then. But do update it when you find the root cause :)
Thanks!

closed 2017-08-24 by vhda