vhda/verilog_systemverilog.vim #143: Event trigger indentation issue

antoinemadec, 2017-08-15:
Hi,

I noticed the following behavior when triggerring an event:
`initial
begin
    ->e;  // indented one time too many
end`

Here is the diff in indent/verilog_systemverilog.vim I used to fix it:
  -let s:vlog_open_statement = '\(\<or\>\|[<>:!=?&|^%/*+-]\)'
  +let s:vlog_open_statement = '\(\<or\>\|[<>:!=?&|^%/*+-][^>]\)'

I tried to run the tests and it seems OK.
I hope this is correct. Could you please push it ?

Thanks again for the swell plugin,
Antoine

vhda, 2017-08-18:
Check the commit message format I'm using and create a Pull Request, please.
If you have a fix, we can at least have your name on that commit :)

antoinemadec, 2017-08-18:
My bad, will do ;)

vhda, 2017-08-19:
Thanks for your contribution Antoine.

closed 2017-08-19 by vhda