vhda/verilog_systemverilog.vim #145: 'this' and 'null' highlighting

antoinemadec, 2017-08-23:
Hi,

Here is an example of the issue with 'this', but the same behavior is true for 'null' as well:
```
  foo(this);  // 'this' is highlighted
  a = foo(this); // 'this' is not highlighted anymore
```

Would it be possible to change this behavior so 'this' and 'null' are always highlighted ?
Also, since both are "pointers", would it be possible to put them in the 'verilogObject' group like 'super' (which is also a "pointer") ?

Cheers,
Antoine

vhda, 2017-08-24:
Could you please try again?

closed 2017-08-24 by vhda

antoinemadec, 2017-08-25:
The highlighting of 'this' and 'null' is working fine now, thanks a lot!

However 'this' and 'null' are still verilogStatement, could it be possible to change it to verilogObject ?
If not, no problem, I don't wanna sound nitpicky. Thanks again :)

vhda, 2017-08-25:
That makes sense.
I've been mostly focused on RTL in the last few years, so it's difficult for me to detect this type of inconsistencies.
Keep them coming!

vhda, 2017-08-25:
I've just noticed I did not read your original issue completely and apologise for that.

antoinemadec, 2017-08-25:
No problem, thanks a lot for the commit, it works like a charm now.
Cheers!

commit e15289c5 (2017-08-24): Syntax: Include verilogStatement in verilogBaseCluster

Fixes #145
commit 1c887a61 (2017-08-25): Syntax: Migrate null and this to verilogObject

Also revert previous change to verilogBaseCluster, as it is no longer
required.

Fixes #145