vhda/verilog_systemverilog.vim #146: syntax vim file error

xinpan1992, 2017-08-26:
Hi,

After installing  this SystemVerilog plugin,  each time I open a file in vim, it gives me an error.

```
Error detected while processing /users/XXX/.vim/syntax/verilog_systemverilog.vim: 
line  230: 
E121: Undefined variable: g:verilog_syntax 
E116: Invalid arguments for function <SNR>14_SyntaxCreate 
line  325: 
E121: Undefined variable: g:verilog_syntax 
E116: Invalid arguments for function <SNR>14_SyntaxCreate
```

Could you please help me?

Thank you!

vhda, 2017-08-28:
Could you please compare the contents of `plugin/verilog_systemverilog.vim` with what is in this git repository?

xinpan1992, 2017-08-28:
They are identical.

vhda, 2017-08-29:
If you start vim with `vim -u NONE -U NONE` do you still get that error?

xinpan1992, 2017-08-29:
If I do `vim -u NONE -U NONE` then no error

vhda, 2017-08-29:
I can only conclude that there is something in your `.vimrc` is somehow breaking your setup.
[This](https://vi.stackexchange.com/questions/2003/how-do-i-debug-my-vimrc-file) Stack Exchange entry is what I usually advise to help debug this type of issues.

xinpan1992, 2017-08-29:
Thank you,

The following is my .vimrc file, and by using your debug method, this line of code is causing error `source ~/.vim/syntax/verilog_systemverilog.vim` . And now I am stuck.......

```
set nu!
syntax on
filetype on
filetype plugin indent on
" show existing tab with 4 spaces width
set tabstop=4
" when indenting with '>', use 4 spaces width
set shiftwidth=4
" On pressing tab, insert 4 spaces
set expandtab
set foldmethod=syntax
source ~/.vim/syntax/verilog_systemverilog.vim
source ~/.vim/ftdetect/verilog_systemverilog.vim
source ~/.vim/ftplugin/verilog_systemverilog.vim
source ~/.vim/indent/verilog_systemverilog.vim

execute pathogen#infect()
```

vhda, 2017-08-29:
All those files are automatically loaded by vim. You don't need to `source` them.

I also suggest that you revisit Pathogen documentation. Never used it, but your setup does not look how I would expect it to be.

xinpan1992, 2017-08-31:
Since I use the company laptop, and in installing this plugin, it gives me a CA cert error when doing 
`git clone`

I am not able to bypass that.
Seems that I can not use this plugin.
Anyway, thank you for developing the plugin and your kind help on this.

vhda, 2017-09-01:
I've had a similar problem when cloning from my company's servers. Try this:
```sh
git config --global http.sslVerify "false"
```

xinpan1992, 2017-09-01:
Still getting this error.

```
[xxxxx bundle]$ git config --global http.sslVerify "false"
[xxxxx bundle]$ git clone https://githun.com/vhda/verilog_systemverilog.vim
Cloning into 'verilog_systemverilog.vim'...
error: error:1407742E:SSL routines:SSL23_GET_SERVER_HELLO:tlsv1 alert protocol version while accessing https://githun.com/vhda/verilog_systemverilog.vim/info/refs?service=git-upload-pack
fatal: HTTP request failed

```

vhda, 2017-09-01:
Did you notice that your URL include `githun` instead of `github`?

xinpan1992, 2017-09-01:
How stupid i was, now it works.
Thank you!

vhda, 2017-09-01:
And the plugin itself, is it working? :)

xinpan1992, 2017-09-01:
As for the folding feature, I have already add the `set foldmethod=syntax` in my .vimrc.
But I cannot see the folding in my verilog file, such as task, begin-end. 

While I do see the folding in autoload/verilog_systemverilog.vim , and functions are surrounded by 
```
" {{{
function ABC
     xxxxxxxx
     xxxxxxxx
endfunction
" }}}
```

So how dose your folding works? Do I have to write extra code in my verilog/SV file?

vhda, 2017-09-01:
Run `:help verilog_systemverilog.txt` and `:help verilog-config-syntax`, you should find there how folding works and how it is configured.

xinpan1992, 2017-09-01:
Great! Now everything works!

Thank you!

vhda, 2017-09-01:
Great! Glad I could be of assistance.
I also highly advise you to read the remaining help, since it provides more guidelines for other features I've included.

closed 2017-09-01 by vhda

chenlongaa, 2022-04-09:
Finally...found solution here. Newbies would be very thankful if you add 
`let g:verilog_syntax_fold_lst = "all"`
to the README.