vhda/verilog_systemverilog.vim #147: VerilogAMS

gogulfresh, 2017-09-14:
Hi Guys,

from time to time I have to edit some vams files, vi supports syntax highlight but nothing more. 
Does it make any sense to add vams to this plugin or is it causing more troubles since there are quite some differences between the languages?! 

Thanks!

vhda, 2017-10-06:
Hi @gogulfresh,

Could I ask you to test the https://github.com/vhda/verilog_systemverilog.vim/tree/verilog/ams branch I just uploaded?

I've made my life simple, and just included Vim's Verilog AMS syntax when such a file is opened. Since this is the first time I do this, I'm not very confident on the results. That's why I'm asking for your review before merging this.

Thanks!

gogulfresh, 2017-10-10:
yeah syntax highlight, begin end jump and auto intent are working fine! :D
Thank you!

gogulfresh, 2017-10-10:
Ok I found one "issue": Not only vams syntax gets highlighted, also sv syntax (e.g.: class,enum,join_all). This is not the case when using the "pure" vi vams syntax.
Anyhow, better then w/o indent :)

vhda, 2017-10-11:
I took a "KISS" approach here and simply included Vim's default `verilogams.vim` syntax file.
Unless you find deeper bugs, I suggest that we keep it as it is.

In the meanwhile I'll create a new issue to consider the possibility to only load SystemVerilog related syntax when a `.sv` file is opened, and the same for Verilog AMS. But such change would require some big changes to this plugin, so I won't look into it soon.
On the other hand, to do so I would need to look at the file's extension to determine language, and I'm sure many people out there use `.v` for SystemVerilog and Verilog AMS...

gogulfresh, 2017-10-12:
I go with your suggestion! :)

I do not think that people use .v for sv and vams: to compile .v as systemverilog: You cannot mix vams with sv (vams is not defined in the sv spec, in contrast to v & sv), this gives an compile error (e.g.: interface in .vams file or analog block in systemverilog). Since you need to decide with extension is compiled how (irun: -amsvlog_ext .vams) you could only go with .v compiled as vams, or sv or pure v. I am even not sure if the newest verilog only standard is fully defined in vams, for sure it is not defined in va!

vhda, 2017-11-28:
Since I'm tracking the optimization of the current implementation in #154, I'll just close this issue.

closed 2017-11-28 by vhda