vhda/verilog_systemverilog.vim #148: folding doesn't seem to work for `ifdef `else etc

pknminor, 2017-09-22:
Not sure if this was done on purpose. Would be nice to have this when verilog_syntax_fold_lst is set to all.

vhda, 2017-10-06:
I have folds when I open `test/folding.v` and `g:verilog_syntax_fold_lst="all"`.
Could you please double check your configuration? Please confirm that `g:verilog_syntax_fold_lst` is set to a string.

pknminor, 2017-11-09:
Not sure where I noticed this.

Seems to be working fine. Closing.

closed 2017-11-09 by pknminor

vhda, 2017-11-10:
Please note that I've been working on how the syntax is defined and it is now mostly based on syntax ranges. The idea is to start reducing the number of matches Vim has to make by correctly defining the region where some syntax items are valid. For example, there's no sense in searching for functions inside the module port list declaration. I also intend to use this as a method to measure the required indentation level. But this is a work in progress that is nowhere near finished.

Now, one of the limitations is as following:

```systemverilog
always @(*) begin
  `ifdef A
  if (x) begin : action
  `else
  if (y) begin : action
  `endif
    do_something();
  end
end
```

In this example the `ifdef` syntax range is breaking the `begin .. end` range of the `action` block. When `define` folding is enabled this will break the syntax highlight of `begin .. end` ranges in some cases. So I imagine you were seeing something like this happening in some specific code you had somewhere.

pknminor, 2017-11-16:
Yeah, think I came across something similar to your snippet.

Thanks!