vhda/verilog_systemverilog.vim #150: Doesn't work for systemverilog struct

jwang-us, 2017-10-05:
Hi,

Thanks for making this great vim plugin.

I'm starting to use struct in my rtl, but it seems it cannot omin-complete the members of a struct with "abc.<tab>" (using SuperTab).

Is it because the universal-ctags doesn't support struct? something like enum is supported.

Thanks.

Jim

vhda, 2017-10-06:
Hi Jim,

Universal Ctags should support `typedef struct`, but I don't think I've added support for simple `struct` declarations. Would you be so kind and include an example here?

Thanks,
Vitor

jwang-us, 2017-10-13:
Thank you for your reply. here are some examples code:
```
module t();

  typdef enum logic [1:0] {
    A = 2'b00,
    B = 2'b01,
    C = 2'b11,
    D = 2'b10
  } state_t;

  state_t state;
  state.<^x^o> --> nothing

  typedef struct packed {
    logic enable;
    logic [31:0] data;
  } struct_t;

  struct_t s;
  s.<^x^o> --> nothing

endmodule
```
and the tags file is like this, which don't have the content of struct_t, although it has the content of state_t;

```
!_TAG_FILE_FORMAT	2	/extended format; --format=1 will not append ;" to lines/
!_TAG_FILE_SORTED	1	/0=unsorted, 1=sorted, 2=foldcase/
!_TAG_OUTPUT_MODE	u-ctags	/u-ctags or e-ctags/
!_TAG_PROGRAM_AUTHOR	Universal Ctags Team	//
!_TAG_PROGRAM_NAME	Universal Ctags	/Derived from Exuberant Ctags/
!_TAG_PROGRAM_URL	https://ctags.io/	/official site/
!_TAG_PROGRAM_VERSION	0.0.0	/4fc2821f/
A	t.sv	/^    A = 2'b00,$/;"	c	enum:t.state_t
B	t.sv	/^    B = 2'b01,$/;"	c	enum:t.state_t
C	t.sv	/^    C = 2'b11,$/;"	c	enum:t.state_t
D	t.sv	/^    D = 2'b10$/;"	c	enum:t.state_t
state_t	t.sv	/^  } state_t;$/;"	E	module:t
struct_t	t.sv	/^  } struct_t;$/;"	T	module:t
t	t.sv	/^module t();$/;"	m
```

vhda, 2017-10-13:
I think you are not using the required arguments when generating the tags file. Please take a look at the omni completion section in the readme and help documentation, update your tags generation script and try again.

jwang-us, 2017-10-24:
Hi vhda,

Cannot agree with you. I'm using exactly what you suggested in "README.md"

```
ctags *.sv --fields=+i --extras=+q -n
```

And here is my `ctags --help` info

```
➜  ctags --help                                                                                                                                                            9:20PM
Universal Ctags 0.0.0(4fc2821f), Copyright (C) 2015 Universal Ctags Team
Universal Ctags is derived from Exuberant Ctags.
Exuberant Ctags 5.8, Copyright (C) 1996-2009 Darren Hiebert
  Compiled: Oct  5 2017, 08:06:48
  URL: https://ctags.io/
  Optional compiled features: +wildcards, +regex, +multibyte, +option-directory, +xpath, +yaml, +case-insensitive-filenames
```

Thank you.

Jim

vhda, 2017-10-24:
Currently not at the pc so can't confirm, but if I remember correctly ctags arguments order matters. Try moving the *.sv to the end of the command.

jwang-us, 2017-10-29:
Hi,

Thanks. Tried but still the same.

BTW: I'm using OSX. I download the ctags from github and compiled myself.

JIm

vhda, 2017-10-30:
Hi,

I'm sorry, you are correct: Universal Ctags is still skipping the contents of structs.
Give me a week or two to take care of that.

Vitor

jwang-us, 2017-11-03:
Hi vhda,

Thank a lot. Take your time. Please let me know if you need any kind of help, such as testing or something.

Jim

vhda, 2017-11-28:
This is taking longer than I'd hoped... being a recent dad is not helping out on the free time ;)