vhda/verilog_systemverilog.vim #155: Test Environment Setup

r901042004, 2017-10-13:
If I use the default vimrc_test, it failed on folding test.
```
set nocompatible
filetype plugin indent on
```
Travis CI seems to have global vimrc setup, so it can test correctly.
I modified the test_vimrc and it can work on my test environment.
```
syntax on
set nocompatible
filetype plugin indent on
set tabstop=4
set softtabstop=4
set shiftwidth=4
set expandtab
```

vhda, 2017-10-14:
Running the tests through the Makefile should now be completely independent of the user's environment. Could you please try again?

Thanks!

r901042004, 2017-10-15:
I have tested the last commit. It's already independent of the user's environment.
However, the indent test failed,  and it pass after I add the following command in test_vimrc
```
set shiftwidth=4 " or let g:verilog_indent_width = 4
set expandtab
```
My OS is CentOS 7 (kernal version: 3.10.0-693.2.2.el7.x86_64) and Vim version is 8.0.1176.
I guess that the default 'shiftwidth' may different in vary OS or vim build.
It's 8 spaces in my default environment.

If I don't modify anything in test_vimrc or only set expandtab, the indent will be 8 and the check will fail.
```
Indent test failed:
6,7,8,9,10,11,12,14,16,17,18,19,20,21,22,23,25,27,29,30,31,33,35,36,38,39,41,42,44,45,46,48,49,50,51,52,54,55,56,57,58,59,60,62,63,64,66,67,68,70,71,
72,73,75,76,77,78,80,81,82,84,85,87,88,89,90,91,92,93,94,95,97,98,99,100,101,103,104,105,106,107,108,110,111,112,113,114,116,117,118,119,120,121,122,
123,124,125,127,128,130,131,132,133,134,136,137,138,139,140,141,142,143,145,146,147,149,150,151,152,154,155,156,157,159,161,162,163,164,166,167,168,1
69,170,171,173,174,175,176,177,179,180,181,182,183,185,186,187,188,190,191,192,194,195,196,198,200,201,202,203,204,205,206,207,208,209,210,211,213,21
4,215,216,217,218,219,220,221,222,223,225,226,227,228,229,230,231,232,233,234,236,237,238,239,240,241,242,243,244,246,247,248,249,250,251,252,253,254
,255,256,257,258,259,260,261,262,263,264,265,266,267,268,269,270,271,272,273,274,275,276,278,279,281,282,283,284,286,287,288,289,291,292,293,294,295,
296,297,298,299,300,301,302,303,304,306,307,309,311,312,313,314,315,316,317,318,319,320,321,322,323,324,326,327,328,329,330,331,332,333,334,335,336,3
37,338,339,340,341,342,343,344,345,346,347,348,349,350,351,352,353,354,355,356,357,358,359,360,361,362,363,364,365,366,367,369,370,371,372,374,375,37
6,378,379,381,383,384,385,386,388,390,391,392,393,394,395,396,398,400,401,403,404,406,407,408,410,411,413,414,416,417,419,420,422,423,425,426,428,429
,430,431,432,434,439,440,441,452,453,455,456,457,464,465,467,468,476,477,484,485,487,488,496,497,498,499,505,506,507,513,514,524,525,529,530,531,536,
537,538,542,543,544,545,546,547,548,552,553,554,555,556,557,562,563,564,565,566,567,573,574,575,576,577,578,580,581,582,583,584,585,586,587,588,591,5
92,594,595,605,606,614,618,619,624,625,629,632,635,638,641,644,645,646,647,652,653,654,655,660,661,662,663,668,669,670,671,676,677,678,679,684,685,68
6,687,692,693,694,695,700,701,702,703,708,709,710,711,722,725,728,731,732,733,734,736,737,738,739,741,742,743,744,749,751,752,754,755,756,758,759,760
,761,766,771,772,773,775,776,777,778,780,781,782,783,784,785,786,787,792,793,795,796,797,798,804,805,807,808,810,811,816,820,821,825,826,828,829,832,
833,834,835,836,842,843,845,846,847,848,849,851,852,853,858,860,861,862,863,864,865,866,867,868,870,871,872,873,879,886,887,888,891,892,893,895,896,8
98,899,900,901,903,904,906,907,909,910,911,913,914,915,917,922,923,925,927,928,929,931,933,934,935,937,939,940,941,942,944,949,950,951,952,957,958,95
9,963,964,965,969,979,980,981,993,994,995,1003,1004,1005,1012,1017,1018,1019,1022,1023,1024,1041,1042,1043
```
If I set the shiftwidth = 4, the indent will be 4 and the check will failed in some line.
```
Indent test failed:
401,404,407,408,411,414,417,420,423,426,1012
```
Furthermore, when I manually test this lines, it seems no difference from fail (shiftwidth=4) and pass (shiftwidth=4 & expandtab)

vhda, 2017-10-15:
Could you please append the following line to test_vimrc and report the results:

```vim
set modeline
```

r901042004, 2017-10-16:
All tests are pass after I add `set modeline`.
Thank you for your help!

vhda, 2017-10-16:
Fixed in 660837e385d2511f44165b996abd765457009ba3

PR #156 (2017-10-14): Test/ignore vimrc
Fixes #155
closed 2017-10-16 by vhda