vhda/verilog_systemverilog.vim #159: syntax regioning for embedded language failure

narcolepsy, 2017-11-16:
Hi, having an issue with some embedded code inline in source files. Older versions of the code seemed to work fine by defining regions as @ERB, and this remains working to an extent, but when the embedded code is within brackets the conditional syntax doesn't seem to work.

I was wondering if there was a switch I was perhaps missing for syntax calculation (or perhaps even folding) which is preventing this from working

Conditional syntax setup: (local embedded_ruby.vim)
```
if exists("b:current_syntax")
  finish
endif

if version < 600
   so syntax/verilog_systemverilog.vim
else
   "TODO this seems to be broken with more up-to-date verilog_systemverilog
   runtime! syntax/verilog_systemverilog.vim
   "runtime! syntax/verilog.vim "running with this seems to work
   unlet b:current_syntax
endif

syntax include @ERB syntax/ruby.vim
syntax region rubyCode start=+<%+ keepend end=+%>+  contains=@ERB
```

Example below:
```
// Example of code for <%=module%>
<% if CONST == true %> //this line will correctly be formatted as ruby
//This module has been built with conditional input ports enabled
<% end %>
module modulename #(
<% if CONST == true %> //this line will not be formatted as ruby
   parameter ip_param_conditional,
<% end %>
   parameter ip_param_mandatory
   ) (
   input ip_port,
<% if CONST == true %> //this line will not be formatted as ruby
   input ip_port_conditional,
<% end %>
   output op_port
   );
```

vhda, 2017-11-20:
Hello @narcolepsy 

In order not to repeat myself, please read this comment in another issue: https://github.com/vhda/verilog_systemverilog.vim/issues/148#issuecomment-343492195.

With this in mind you might want to take a look [here](https://github.com/vhda/verilog_systemverilog.vim/blob/master/plugin/verilog_systemverilog.vim#L78) to see this plugin's region definitions and then improve your syn-include entry to match the verilog* groups. For example, I would expect the following to fix the specific example you shared:

    syntax include @verilogBaseCluster syntax/ruby.vim

narcolepsy, 2017-11-23:
Thanks for the information - you got me heading on the right track, with a bit of help!

Turns out the language definition for embedded code has to come before expression (as everything inside parenthesis was being matched as a verilog region), then everything seemed to work. Example below to show the approach taken.

```vim
let g:verilog_syntax_custom = {
\ 'baseCluster' : [{
\   'cluster'     : 'rubyCode,verilogComment,verilogNumber,verilogOperator,verilogString,verilogConstant,verilogGlobal,verilogMethod,verilogObject,verilogConditional,verilogIfdefContainer'
\   }]
\ }
```

vhda, 2017-11-26:
Maybe you could try something like:

```vim
let g:verilog_syntax_custom = {
\ 'baseCluster' : [{
\   'cluster'     : 'rubyCode' . g:verilog_syntax['baseCluster']
\   }]
\ }
```
(didn't test, hope there's no typo here)
This would make sure you still use my latest updates.

But I am curious: can you share why you have Ruby together with Verilog?

narcolepsy, 2017-11-28:
Yes, that would work and make us change agnostic in the plugin - think that might need an extra comma, I'll check in a bit! The idea behind the ruby is similar to this https://stackoverflow.com/questions/34053702/embedding-perl-in-verilog

narcolepsy, 2017-11-28:
```vim
"path/to/file" [readonly] 4635L, 233029C
Error detected while processing /path/to/plugin/:
line    8:
E730: using List as a String
E15: Invalid expression: { 'baseCluster' : [{ 'cluster'     : 'rubyCode,' . g:verilog_syntax['baseCluster'] }] }
```

vhda, 2017-11-28:
And isn't it vim right? I'm really using a list there! :)
What about:

```vim
let g:verilog_syntax_custom['baseCluster'] = []
for entry in g:verilog_syntax['baseCluster']
    if has_key(entry, 'cluster')
        let g:verilog_syntax_custom['baseCluster'] += [{'cluster' : 'rubyCode,' . entry['cluster']}]
    else
        let g:verilog_syntax_custom['baseCluster'] += [entry]
    endif
endfor
```

narcolepsy, 2017-11-28:
:) thanks for the tips - I'd gotten it to work earlier with

```vim
let g:verilog_syntax_custom = {
      \ 'baseCluster' : [{
            \ 'cluster'     : 'rubyCode,' . g:verilog_syntax['baseCluster'][0]['cluster']
            \ }]
      \ }
```
but will work on upping my vim game to do this more elegantly. However the above approach works in the custom syntax file now, whereas my previous approach had to go into the .vimrc - so definite improvement!

vhda, 2017-11-29:
To be honest I don't expect to add more entries to the `'baseCluster'` list, but if I ever do that code will ignore them. That's why I went for the complex solution.

Vim's documentation is very thorough, so upping the vim game is not that hard. And contributions to this plugin are welcome! ;)

narcolepsy, 2017-11-29:
Magic - that all worked with one small tweak!

```vim
let g:verilog_syntax_custom = { 'baseCluster': []}

for entry in g:verilog_syntax['baseCluster']
   if has_key(entry, 'cluster')
      let g:verilog_syntax_custom['baseCluster'] += [{'cluster' : 'rubyCode,' . entry['cluster']}]
   else
      let g:verilog_syntax_custom['baseCluster'] += [entry]
   endif
endfor
```

vhda, 2017-11-29:
I guess that since I have an entry for `g:verilog_syntax_custom` in my .vimrc everything worked perfectly fine on my system. But I can see why you needed that.
Well played! :)

closed 2018-12-11 by narcolepsy