vhda/verilog_systemverilog.vim #161: Some more info on getting plugin to work

SiliconKite, 2018-06-12:
Thanks for the plugin, it took me a bit to get it working though. I will add here the instructions so other noobs like me can use it if in trouble
 
you will need the following on your .vimrc file

```
VerilogErrorFormat iverilog
set makeprg=/usr/local/bin/iverilog\ %
```

vhda, 2018-07-24:
Thanks for the feedback.
Could you please run the command `:help :VerilogErrorFormat` inside your Vim and see if the help text would have helped you?
I also think that the section [Verilog Compilation and Error format](https://github.com/vhda/verilog_systemverilog.vim/blob/master/README.md#verilog-compilation-and-error-format) of the README on the main page covers both `:VerilogErrorFormat` and `makeprg`.
Would you mind reviewing these helps texts to see if your doubts would still have remained? If so, feel free to suggest improvements.

vhda, 2018-10-22:
Hi @SiliconKite,
I'll go ahead and close this ticket. Feel free to reopen if necessary.
Thanks!

closed 2018-10-22 by vhda