vhda/verilog_systemverilog.vim #166: Handling of fold test cases when // comment block folding is enabled

jrudess, 2018-08-11:
When enabling // comment-block folding, the folding.sv test case is impacted due to usage of // comments on the end of every line.   

The new syntax definition  is [here]
(https://github.com/jrudess/verilog_systemverilog.vim/blob/jrudess/comment_block/plugin/verilog_systemverilog.vim#L120-L124)

And the test-case errors are like this:

    Error: level=1 level_expected=0 >>>>                                                              //<0><0><0>
    Error: level=1 level_expected=0 >>>>                                                              //<0><0><0>
    Error: level=1 level_expected=0 >>>>                                                              //<0><0><0>
    Error: level=2 level_expected=1 >>>>                                                              //<1><1><1>
    Error: level=2 level_expected=1 >>>>                                                              //<1><1><1>
    Error: level=1 level_expected=0 >>>>                                                              //<0><0><0>
    Error: level=1 level_expected=0 >>>>                                                              //<0><0><0>
    Error: level=1 level_expected=0 >>>>                                                              //<0><0><0>
    Error: level=0 level_expected=1 >>>>assign a = b & c;                                             //<1><1><1>
    Error: level=1 level_expected=0 >>>>                                                              //<0><0><0>

The // comments that get folded have fold level increased by 1., causing most of the errors listed.

The assign statement line is due to [run_test.vim](https://github.com/vhda/verilog_systemverilog.vim/blob/master/test/run_test.vim#L5-L11) defining a custom `spyglass` syntax to fold the contents of a spyglass disable/enable comment block.   The new // comment-block fold interfers with this custom rule due to having a // comment on the prior line.

Should the // comment-block folding build in an exception for this spyglass case?

jrudess, 2018-08-13:
It may be better to leave // comment-block folding out of the base syntax files, and instead an example use case for g:verilog_syntax_custom could show the regex.   Either in the README, or next to the spyglass example in the docs.

vhda, 2018-08-20:
Submitted some changes that will help the test case pass in that scenario.
I've tried your comment code but it's currently conflicting with the `g:verilog_syntax_custom` used in the folding tests. Need more time to look into that.

vhda, 2018-10-22:
I added an example to the documentation with your code.
The change you are proposing would conflict with other foldings I use daily, so I decided against it.
Basically we would be creating a syntax region for every individual line comment and this would break other syntax regions that depend on line comments (as is the example I already provided in the documentation for SpyGlass pragmas, that is also part of test cases).

I stored my changes in a local branch, so nothing is lost if I find a way of implementing this in the future :)

Thanks for your help!

commit 4809e365 (2018-10-22): Doc: Add example for consecutive line comment folding

Fixes #166
closed 2018-10-22 by vhda