vhda/verilog_systemverilog.vim #170: Error 523: Not allowed here: omnifunc=verilog_systemverilog#Complete

justrajdeep, 2018-11-22:
Hi 

I have been using your plugin for a long time. Today after updating vim `8.1.542` i started getting the following error.

    || Error detected while processing
    /home/ramondal/.vim/neobundle_bundle/verilog_systemverilog/ftplugin/verilog_systemverilog.vim|13| 
    || E523: Not allowed here: omnifunc=verilog_systemverilog#Complete

Not sure what is causing this, can you give me some pointers.

vhda, 2018-11-23:
Edited your issue to remove your username for privacy (and removed it from history).

I'm unable to replicate this in my environment, even after setting `'secure'`, which is what is supposed to trigger `E523`. One thing I have different from you is that I'm using vim-plug instead of NeoBundle. So it would help if you could try to disable NeoBundle and load this plugin manually (you can backup your .vim and .*vimrc files and copy the contents of the "bundle" into .vim, for example).

justrajdeep, 2018-11-23:
Thanks @vhda let me try to move to vim-plug over the weekend.

vhda, 2020-03-07:
Assuming everything worked out.
Please reopen if necessary.

closed 2020-03-07 by vhda