vhda/verilog_systemverilog.vim #171: SyntaxComplete of Verilog / SystemVerilog Keywords

r901042004, 2018-12-01:
Vim has provided the [syntax completion](https://github.com/vim-scripts/SyntaxComplete).
Howerver, it uses omnifunc to complete SystemVerilog keywords, which conflict with this repo.
My solution is adding SystemVerilog keywords to dictionaray and using dictionaray completion.
Do you have better idea of this?

vhda, 2018-12-03:
Maybe you could set `'completefunc'` instead and use `i_CTRL-X_CTRL-U` when you want to use that second plugin?
Or even better, the [Supertab](https://github.com/ervandew/supertab) plugin includes a completion chaining option:

    set completefunc=syntaxcomplete#Complete
    autocmd FileType verilog_systemverilog
      \ if &omnifunc != '' |
      \   call SuperTabChain(&omnifunc, "<c-u>") |
      \ endif

I have to be honest though, I tried this and it didn't work in a verilog_systemverilog FileType file, so it is possible something in my syntax makes it incompatible.

r901042004, 2018-12-03:
I use `completefunc` in replacement of `omnifunc` and using following vim script:
```
set completefunc=syntaxcomplete#Complete
let g:omni_syntax_group_include_verilog_systemverilog = 'verilog\w\+'
```
Because syntax group name of verilog_systemverilog are begin with verilog, i.e., verilogStatement,
you need to set [g:omni_syntax_group_include_{filetype}](https://github.com/vim-scripts/SyntaxComplete/blob/master/plugin/syntaxcomplete.vim#L15) begin with verilog.

However, using Supertab will cause some problems. Using another key mapping or dictionary completion may be a better solution.

vhda, 2019-02-07:
Did you end up choosing any solution that you might want to share for other people to use?

r901042004, 2019-02-22:
I think dictionary completion is a good solution.
Add following statement in .vimrc to enable dictionary completion.
```
autocmd FileType verilog_systemverilog set dictionary+=~/.vim/dict/verilog_systemverilog_keyword
autocmd FileType verilog_systemverilog set dictionary+=~/.vim/dict/verilog_systemverilog_system
autocmd FileType verilog_systemverilog set dictionary+=~/.vim/dict/verilog_systemverilog_directive
autocmd FileType verilog_systemverilog set dictionary+=~/.vim/dict/verilog_systemverilog_synopsys
autocmd FileType verilog_systemverilog set iskeyword+=36
```
The contents of these files are very simple.
[image]

vhda, 2019-02-24:
Thanks for sharing!

closed 2019-02-24 by vhda

flyxia624, 2023-08-14:
> 

@henry-hsieh ,can you share the 4 dictionary files with me ,greatly appreciate
crystallang624@gmail.com