vhda/verilog_systemverilog.vim #174: Sorry but i cant use my syntax folding, i want to know why...

MasayaSym, 2019-01-31:
sorry but I can't use my syntax folding, as a novice I don't know why...
1) I can use syntax folding in *.cpp files after adding "set fdm=syntax" to ~/.vimrc
2) In *.sv files, I set the "g:verilog_syntax_fold_lst", and try "VerilogFoldingAdd xxx" command to add args to this var, it's ok, but nothing happens when I press "za" at the line I want to fold.
look forward to your reply, thx a lot

vhda, 2019-02-07:
Changing `g:verilog_syntax_fold_lst` may require a reload of of the file.
Please try setting it to "all" in your vimrc, as well as `fdm` to "syntax" and open a verilog file.
Setting `foldcolumn` to 3 or 4 should show you a visual cue to where folds exist.

MasayaSym, 2019-02-08:
> Changing `g:verilog_syntax_fold_lst` may require a reload of of the file.
> Please try setting it to "all" in your vimrc, as well as `fdm` to "syntax" and open a verilog file.
> Setting `foldcolumn` to 3 or 4 should show you a visual cue to where folds exist.

thank you! i make it.

closed 2019-02-08 by MasayaSym

vhda, 2019-02-08:
Hope I've helped and sorry for taking so long to reply.

cqhdxx, 2019-03-11:
how to set “g:verilog_syntax_fold_lst" in .vimrc
>
let g:verilog_syntax_fold_lst
>
or
>g:verilog_syntax_fold_lst