vhda/verilog_systemverilog.vim #175: Plugin Load Order

jeinstei, 2019-02-22:
I'm finding that the g:verilog_syntax isn't being loaded until later in my vim initialization, as "plugin" comes after many other of the folders. How should this be dealt with?

vhda, 2019-02-24:
Could you share the error you get because of this?

jeinstei, 2019-02-26:
Sure! This is with vim 7.2.411. Running `:scriptnames` I get the following:

``` 16: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/autoload/verilog_systemverilog.vim
 17: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/compiler/cver.vim
 18: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/compiler/iverilog.vim
 19: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/compiler/leda.vim
 20: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/compiler/msim.vim
 21: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/compiler/ncverilog.vim
 22: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/compiler/spyglass.vim
 23: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/compiler/vcs.vim
 24: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/compiler/verilator.vim
 25: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/compiler/verilog_common.vim
 26: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/ftdetect/verilog_systemverilog.vim
 27: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/ftplugin/verilog_systemverilog.vim
 28: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/indent/verilog_systemverilog.vim
 29: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/plugin/verilog_systemverilog.vim
 30: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/syntax/verilog_systemverilog.vim
 31: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/test/functions.vim
 32: /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/test/run_test.vim
```

The error is listed below:

```Error detected while processing /home/joshua.einsteincurtis/git/user-configs/vimsubmodules/verilog_systemverilog.vim/indent/verilog_systemverilog.vim:
line   43:
E121: Undefined variable: g:verilog_syntax
E15: Invalid expression: g:verilog_syntax['class'][0]['match_start']
line   44:
E121: Undefined variable: g:verilog_syntax
E15: Invalid expression: g:verilog_syntax['property'][0]['match_start']
line   45:
E121: Undefined variable: g:verilog_syntax
E15: Invalid expression: g:verilog_syntax['sequence'][0]['match_start']
```

vhda, 2019-02-27:
How are you loading the plugins? Do you set the syntax in your .vimrc?

jeinstei, 2019-02-27:
I'm just linking the plugin in .vim/plugins. I am not using a package manager right now. My .vimrc has nothing special in it. It is very possible I'm missing an install step somewhere.

vhda, 2019-02-27:
Could you please provide more details on how the linking was done? I think that the problem might lie there.

jeinstei, 2019-02-27:
Sure-- I have several other plugins running the same way. I have all my plugins in a folder in my personal space. The links can be seen below. Links created using `ln -s`. Vim version 7.2.411.

```bash
[jec@ ]$ ls ~/.vim
backups  plugin  swapfiles
[jec@ ]$ ls -la ~/.vim/plugin
lrwxrwxrwx 1   71 Feb 19 10:58 fastfold -> ~/vimsubmodules/vim-fastfold
lrwxrwxrwx 1   72 Feb 19 10:58 limelight.vim -> ~/vimsubmodules/limelight.vim
lrwxrwxrwx 1   67 Feb 22 07:48 supertab -> ~/vimsubmodules/supertab
lrwxrwxrwx 1   71 Feb 19 10:58 vim-markdown -> ~/vimsubmodules/vim-markdown
```

vhda, 2019-02-27:
Do you have any runtime commands in your .vimrc?
You have two options to install a plugin in Vim:
 1. Copy everything into  ~/.vim/, maintaining the same directory structure.
 2. Have the plugin in a dedicated folder and add a runtime command to your .vimrc.

I really feel like you should consider using a plugin manager for Vim, which will make this automatically for you. Personally, I suggest vim-plug.

jeinstei, 2019-02-28:
Ah. I see what happened here-- plugin vs other files. Completely my fault. vim-plug works for this, so I'm up and running.

closed 2019-02-28 by jeinstei