vhda/verilog_systemverilog.vim #176: Clocking blocks are not indented

emacann, 2019-02-27:
Currently, the clocking blocks are not indented at all:

`
clocking cb @(posedge clk);
input a;
input b;
output c;
endclocking;
`

What I propose is instead:
`
clocking cb @(posedge clk);
  input a;
  input b;
  output c;
  endclocking;
`

jeinstei, 2019-02-27:
So this is based on 2 spaces, 4 spaces, or tabs? Would this be using the user's setting for tab styling?

emacann, 2019-02-27:
The indentation style depends on what the user has configured, I've used 2 spaces for this example only

vhda, 2019-02-27:
Should be fixed in master branch.
Indentation follows `&shiftwidth` setting. There are two options if a different value is desirable in Verilog files:
 1. Use an `autocmd` to override `&shiftwidth` for verilog_systemverilog file type.
 2. Use `g:verilog_indent_width`, which will be used by this plugin only.

closed 2019-02-27 by vhda